Entware/Entware #739: Problems upgrading SoftEtherVPN server through opkg

Myron-S, 2021-12-22:
Output from opkg upgrade regardless of there being an update or not.

```
root@ZEUS home # opkg upgrade
Configuring softethervpn5-libs.
Creating hamcore.se2 archive...
hamcore.se2 builder

Destination: "/opt/libexec/softethervpn/hamcore"
Source: "/opt/libexec/softethervpn/hamcore.se2"

tinydir_open_sorted() failed!
Collected errors:
 * pkg_run_script: package "softethervpn5-libs" postinst script returned status 1.
 * opkg_configure: softethervpn5-libs.postinst returned 1.
root@ZEUS home #

```
This happens if there is a SoftEtherVPN update or not and I believe that SoftEtherVPN will not upgrade or install using opkg;
I have no clues on how to troubleshoot this issue.

opkg used to upgrade SoftEtherVPN server in the past.

Environment:
 - Entware feed: x64-k3.2
 - Device: Western Digital DL4100 NAS
 - Firmware version: OS5

Myron-S, 2021-12-22:
I think I found the problem within /opt/lib/opkg/info/softethervpn5-libs.postinst.

Script is:
```
#!/bin/sh

echo 'Creating hamcore.se2 archive... '
GCONV_PATH=/opt/lib/gconv hamcorebuilder /opt/libexec/softethervpn/hamcore /opt/libexec/softethervpn/hamcore.se2 && rm -fr /opt/libexec/softethervpn/hamcore

```
What worked is:
```
root@ZEUS info # cat ./softethervpn5-libs.postinst
#!/bin/sh

echo 'Creating hamcore.se2 archive... '
GCONV_PATH=/opt/lib/gconv hamcorebuilder /opt/libexec/softethervpn/hamcore.se2 /opt/libexec/softethervpn/hamcore && rm -fr /opt/libexec/softethervpn/hamcore

```
Now, can anyone tell me if my solution is correct or have I just made more problems for the future?

closed 2021-12-22 by The-BB

ryzhovau, 2021-12-22:
Thanks for report! You're right, `hamcorebuilder` usage syntax was changed recently. Please, update package and give a feedback

Myron-S, 2021-12-22:
The result . . .
```
root@ZEUS home # opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
root@ZEUS home # opkg upgrade
Upgrading softethervpn5-libs on root from 5.02.5180-2 to 5.02.5180-2a...
Downloading http://bin.entware.net/x64-k3.2/softethervpn5-libs_5.02.5180-2a_x64-3.2.ipk
Deleting hamcore.se2 archive...
Upgrading softethervpn5-server on root from 5.02.5180-2 to 5.02.5180-2a...
Downloading http://bin.entware.net/x64-k3.2/softethervpn5-server_5.02.5180-2a_x64-3.2.ipk
Configuring softethervpn5-libs.
Creating hamcore.se2 archive...
hamcore.se2 builder

Destination: "/opt/libexec/softethervpn/hamcore.se2"
Source: "/opt/libexec/softethervpn/hamcore"

Done!
Configuring softethervpn5-server.
Collected errors:
 * resolve_conffiles: Existing conffile /opt/libexec/softethervpn/vpn_server.config is different from the conffile in the new package. The new conffile will be placed at /opt/libexec/softethervpn/vpn_server.config-opkg.
root@ZEUS home #

```
Works as expected. Thank-you.