Entware/Entware #740: please add package Judy

kimdoyeonsa, 2021-12-25:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

please add package Judy
https://sourceforge.net/projects/judy/

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-12-25:
Sorry, no plans to add it.

* abandoned 10+ years ago,
* it's a library with no any app linked against.

closed 2021-12-25 by ryzhovau

kimdoyeonsa, 2021-12-25:
please add package libjudy
https://github.com/netdata/libjudy