Entware/Entware #741: update sslh

tleepa, 2021-12-25:
**Describe the solution you'd like**
Can I ask you to update sslh (https://github.com/yrutschle/sslh/)?

Version 1.21c (sslh) does not work with libconfing 1.7.3 because of some bug that was patched later.
The current, latest tag for sslh is 1.22c. The above mentioned bug was fixed in 1.22.

fett327, 2022-01-22:
Second this request, for the exact same reason the OP mentioned.

fett327, 2022-03-30:
FYI, I finally got around to updating this package within the Openwrt repo ([PR#18164](https://github.com/openwrt/packages/pull/18164)).
@ryzhovau Happy to wait for the next standard sync, but I can also make a pull request if that helps you guys.  I already tested on armv7-3.2

fett327, 2022-04-02:
Pull request created
https://github.com/Entware/entware-packages/pull/22

ryzhovau, 2022-04-15:
Done.

closed 2022-04-15 by ryzhovau