Entware/Entware #742: Go patch missing SSL directory

maforshaw, 2021-12-25:
After upgrading Entware recently I noticed rclone was not working any more with my custom CA certificate. From what I can tell looking at the Entware patch, a section appears to have gone missing from:

Entware/tools/go-src/patches/500-entware.patch

All previous commits until 711be44 had the following:

```
--- a/src/crypto/x509/root_unix.go
+++ b/src/crypto/x509/root_unix.go
@@ -16,6 +16,7 @@ import (
 // Possible directories with certificate files; stop after successfully
 // reading at least one file from a directory.
 var certDirectories = []string{
+	"/opt/etc/ssl/certs",           // Entware
 	"/etc/ssl/certs",               // SLES10/SLES11, https://golang.org/issue/12139
 	"/system/etc/security/cacerts", // Android
 	"/usr/local/share/certs",       // FreeBSD
```

That block appears to be now missing on the master branch which means folder "/opt/etc/ssl/certs" is no longer searched.

Looking at the Go source code, it appears that block moved from file "root_unix.go" to file "root_linux.go" so it looks like the Entware patch was accidentality removed instead of patching "root_linux.go" along with the other change already made in that file?

zyxmon, 2021-12-25:
The patch is still present - https://github.com/Entware/Entware/blob/master/tools/go-src/patches/500-entware.patch#L7

```
--- a/src/crypto/x509/root_linux.go
+++ b/src/crypto/x509/root_linux.go
@@ -6,6 +6,7 @@ package x509

 // Possible certificate files; stop after finding one.
 var certFiles = []string{
+       "/opt/etc/ssl/certs/ca-certificates.crt",            // Entware
        "/etc/ssl/certs/ca-certificates.crt",                // Debian/Ubuntu/Gentoo etc.
        "/etc/pki/tls/certs/ca-bundle.crt",                  // Fedora/RHEL 6
        "/etc/ssl/ca-bundle.pem",                            // OpenSUSE

```
File `src/crypto/x509/root_unix.go` does not have string constants for certificates directory. The comments in this file state that in unix there are SSL_CERT_DIR and SSL_CERT_FILE environment variables that define certificates.

Please describe rclone problem with custom certificates with more details.

maforshaw, 2021-12-25:
No, that patch is for **file** "/opt/etc/ssl/certs/ca-certificates.crt" not the **directory** "/opt/etc/ssl/certs". Like I said, the Go source has changed. Please see:

https://github.com/golang/go/blob/master/src/crypto/x509/root_linux.go

As you can see (at the bottom), they have moved the directory check into "root_linux.go". Previously it was in "root_unix.go".

By removing the Entware patch, CA certificates files in **directory** "/opt/etc/ssl/certs" are now ignored where previously they were not.

rclone now gives an invalid CA certificate error because it cannot find the CA certificate.

zyxmon, 2021-12-26:
Thanks - that is missing. This is fixed now by - https://github.com/Entware/Entware/commit/8898065793e63080b6dcda1dcf800ccfad4930f7
What is yor architecture - I'll rebuild rclone for you to test that the bug is fixed.

maforshaw, 2021-12-26:
Great, thank you. Obviously there is a work around by manually appending my CA to the ca-certificates.crt file but that will get overwritten by the next update so that is why I reported the issue.

My architecture is armv7, from here: http://bin.entware.net/armv7sf-k3.2

zyxmon, 2021-12-26:
Here are two updated packages - http://test.entware.net/armv7sf-k3.2/  (normal and no hrd float).
Please test and report.
`opkg install <URL>`

maforshaw, 2021-12-26:
I have just installed the test build and it now works perfectly, thank you. This issue can be closed as fixed.

ryzhovau, 2021-12-26:
Thanks for feedback!

closed 2021-12-26 by ryzhovau