Entware/Entware #747: please add package gcc-8.5.0

kimdoyeonsa, 2021-12-28:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

https://github.com/kimdoyeonsa/kdy/tree/main/gcc/patches
https://github.com/kimdoyeonsa/kdy/blob/main/gcc/Makefile
please add package gcc-8.5.0

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-12-28:
No plans to develop native compilation tools, sorry.
The existing tool set [provided](https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation) "as is" without any further development.

closed 2021-12-28 by ryzhovau

ryzhovau, 2021-12-29:
BTW, this package came from upstream with no changes. It's better to send PR to OpenWrt guys.
https://github.com/openwrt/packages/tree/master/devel/gcc