Entware/Entware #750: Issue installing entware

wellloaded, 2022-01-03:
Attempting this on FreshTomato. I've used the internal entware_install.sh script.

```
root@sparrow:/tmp/mnt/usb# entware-install.sh
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Warning: Folder /opt/bin exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/etc exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/lib/opkg exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/tmp exists!
Warning: If something goes wrong please clean /opt folder and try again.
Warning: Folder /opt/var/lock exists!
Warning: If something goes wrong please clean /opt folder and try again.
Info: Opkg package manager deployment...
Connecting to bin.entware.net (104.21.91.83:443)
saving to '/opt/bin/opkg'
opkg                 100% |*********************************************************************************************************************************************************************************************************************************************************|  747k  0:00:00 ETA
'/opt/bin/opkg' saved
Connecting to bin.entware.net (104.21.91.83:443)
saving to '/opt/etc/opkg.conf'
opkg.conf            100% |*********************************************************************************************************************************************************************************************************************************************************|   171  0:00:00 ETA
'/opt/etc/opkg.conf' saved
Connecting to bin.entware.net (104.21.91.83:443)
saving to '/opt/lib/ld-2.23.so'
ld-2.23.so           100% |*********************************************************************************************************************************************************************************************************************************************************|  139k  0:00:00 ETA
'/opt/lib/ld-2.23.so' saved
Connecting to bin.entware.net (104.21.91.83:443)
saving to '/opt/lib/libc-2.23.so'
libc-2.23.so         100% |*********************************************************************************************************************************************************************************************************************************************************| 1190k  0:00:00 ETA
'/opt/lib/libc-2.23.so' saved
Connecting to bin.entware.net (104.21.91.83:443)
saving to '/opt/lib/libgcc_s.so.1'
libgcc_s.so.1        100% |*********************************************************************************************************************************************************************************************************************************************************| 44456  0:00:00 ETA
'/opt/lib/libgcc_s.so.1' saved
Connecting to bin.entware.net (104.21.91.83:443)
saving to '/opt/lib/libpthread-2.23.so'
libpthread-2.23.so   100% |*********************************************************************************************************************************************************************************************************************************************************| 92648  0:00:00 ETA
'/opt/lib/libpthread-2.23.so' saved
ln: ld-linux.so.3: Operation not permitted
ln: libc.so.6: Operation not permitted
ln: libpthread.so.0: Operation not permitted
Info: Basic packages installation...
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Package entware-opt (227000-3) installed in root is up to date.
Configuring entware-opt.
ln: /opt/sbin/ifconfig: Operation not permitted
ln: /opt/sbin/route: Operation not permitted
ln: /opt/bin/netstat: Operation not permitted
ln: /opt/bin/sh: Operation not permitted
ln: /opt/bin/ash: Operation not permitted
Collected errors:
 * pkg_run_script: package "entware-opt" postinst script returned status 1.
 * opkg_configure: entware-opt.postinst returned 1.
ln: /opt/etc/passwd: Operation not permitted
ln: /opt/etc/group: Operation not permitted
ln: /opt/etc/shadow: Operation not permitted
ln: /opt/etc/gshadow: Operation not permitted
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
```

Lots of warnings... If I now try to add a package I get this error:

```
root@sparrow:/tmp/home/root# opkg install mc
Package mc (4.8.27-2) installed in root is up to date.
Configuring entware-opt.
ln: /opt/sbin/ifconfig: Operation not permitted
ln: /opt/sbin/route: Operation not permitted
ln: /opt/bin/netstat: Operation not permitted
ln: /opt/bin/sh: Operation not permitted
ln: /opt/bin/ash: Operation not permitted
Collected errors:
 * pkg_run_script: package "entware-opt" postinst script returned status 1.
 * opkg_configure: entware-opt.postinst returned 1.
```

Am I doing anything wrong?
Thanks

ryzhovau, 2022-01-03:
We do not provide personal support here, sorry.

closed 2022-01-03 by ryzhovau

zyxmon, 2022-01-04:
@wellloaded - it looks like the file system where /opt is located does not support symlinks.