Entware/Entware #752: mktorrent doesn't work with filesize larger than 2GiB

wyyqyl, 2022-01-05:
**Describe the bug**
[mktorrent]: It doesn't work with filesize larger than 2GiB

```
mktorrent -v -p -l 24 -a http://exampletracker.com/announce -o ../test.torrent test.mkv
mktorrent 1.1 (c) 2007, 2009 Emil Renner Berthing

Options:
  Announce URLs:
    1 : http://exampletracker.com/announce
  Torrent name: test.mkv
  Metafile:     ../test.torrent
  Piece length: 16777216
  Be verbose:   yes
  Write date:   yes
  Web Seed URL: none
  Comment:      none

-2109759487 bytes in all.
That's 131 pieces of 16777216 bytes each.

Hashing test.mkv.
Writing metainfo file... done.
```
The exact size of test.mkv is 2185207809 bytes. The length of the file overflows with signed int.

version: `mktorrent 1.1 (c) 2007, 2009 Emil Renner Berthing`

**Environment (please complete the following information):**
 - Entware feed: [x64-k3.2]
 - Device [Synology DS3617xs]
 - Firmware version [DSM 6.2.2-24922 Update 4]

zyxmon, 2022-01-07:
Note: May be this will help - https://github.com/pobrn/mktorrent/issues/28#issuecomment-439714192

wyyqyl, 2022-01-07:
> Note: May be this will help - [pobrn/mktorrent#28 (comment)](https://github.com/pobrn/mktorrent/issues/28#issuecomment-439714192)

Thanks for the information.

```
# This is needed on certain 32bit OSes (notably 32bit Linux) to support
# files and torrents > 2Gb.
#USE_LARGE_FILES = 1
```
My OS is 64bit, not 32bit. According to the comment, it works with 32bit OS.

closed 2022-04-15 by ryzhovau