Entware/Entware #775: Installing Entware without /opt

Otus9051, 2022-03-15:
Hey!
I have a router from my ISP which doesn't have `/opt` and `/` is write-protected. However, `etc` `tmp` `flash` & `flash2` (ISP Specific) are writable. The Router has an EcoNet EN7528 SOC with MIPS 1004Kc V2.15 and Linux 3.18.21. It has 500MB of RAM, and possibly 40-50mb of storage however I am not really sure. Is there anyway to install Entware to any of those directories? The firmware can be found here https://github.com/itsyourap/JioFiber-Home-Gateway/tree/master/Firmwares/JCOW414%20and%20JCOW4032/2.30

Otus9051, 2022-03-15:
Entware is hardlinked to /opt
Dumb decision indeed.

zyxmon, 2022-03-15:
Entware can be installed in chroot environment as described here (Russian) - http://forums.zyxmon.org/viewtopic.php?f=5&t=5797
In case you need statically linkede busybox for your arch - I can upload it.

Otus9051, 2022-03-15:
huh i guess that could work

Otus9051, 2022-03-17:
Get an EXT4 Formatted USB Storage device, mount it to your desired space in the router. Then, download http://bin.entware.net/armv7sf-k3.2/installer/chroot/other/mipsel/install_entware_chroot.sh (MIPSEL only), make a folder on the storage device called `bin` and get http://bin.entware.net/armv7sf-k3.2/installer/chroot/other/mipsel/busybox in that folder. Next, run the `install_entware_chroot.sh` and chroot in with 
```
PATH=/opt/bin:/opt/sbin:/bin:/sbin ./bin/busybox chroot . /bin/sh
```
Then install Entware with the generic.sh script.

closed 2022-03-17 by Otus9051

zyxmon, 2022-03-17:
>> Then install Entware with the generic.sh script.

For chroot install it is preferable to use alternative.sh script