Entware/Entware #778: Kea - Modern DHCP server

FallingSnow, 2022-03-21:
> ISC distributes and maintains TWO full-featured, open source, standards-based DHCP server distributions: Kea DHCP and ISC DHCP. Kea includes all the most-requested features, is far newer, and is designed for a more modern network environment.
https://www.isc.org/kea/

- [ ] No similar package already added and/or this one got some new killer feature(s),
  - There is dnsmasq and isc-dhcp-server-ipv4/6
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2022-04-15:
It uses MySQL or PostgreSQL as backend, which is too heavy for most of embedded devices.
No plans to add it, sorry.

FallingSnow, 2022-04-15:
MySQL and PostgreSQL are optional and not the default. Memfile (a csv on disk) is the default.
> Kea stores its leases in a lease database. The software has been written in a way that makes it possible to choose which database product should be used to store the lease information. Kea supports four database backends: MySQL, PostgreSQL and memfile. To limit external dependencies, MySQL and PostgreSQL support are disabled by default and only memfile is available.

https://kea.readthedocs.io/en/latest/arm/install.html#dhcp-database-installation-and-configuration
> The memfile backend is able to store lease information, but cannot store host reservation details; these must be stored in the configuration file. (There are no plans to add a host reservations storage capability to this backend.)
No special initialization steps are necessary for the memfile backend. During the first run, both kea-dhcp4 and kea-dhcp6 create an empty lease file if one is not present. Necessary disk-write permission is required.

https://kea.readthedocs.io/en/latest/arm/admin.html#supported-backends

ryzhovau, 2022-04-17:
Good! I promise to look into.

closed 2022-07-21 by The-BB