Entware/Entware #782: The python3 version is wrong in the opkg repository

bluzye, 2022-04-10:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - aarch64-3.10
 - Device [e.g. Asus RT-AX88U router]
 - Asus RT-AC86U
 - Firmware version [e.g. asuswrt-merlin]
 -  asuswrt-merlin [386.3_2]

opkg install python3 python3-pip

Then I found that python3 -V is 3.9, but pip is for python 3.10.

ryzhovau, 2022-04-15:
Works for me. Can't reproduce that.
```
alpine86:~# python3 -V
Python 3.10.4
alpine86:~# pip3 -V
pip 22.0.4 from /opt/lib/python3.10/site-packages/pip (python 3.10)

```

closed 2022-04-15 by ryzhovau

bluzye, 2022-04-19:
I have fixed it using below command：
opkg remove --force-depends --force-removal-of-dependent-packages python3-*
opkg remove --force-depends --force-removal-of-dependent-packages python3 python3-pip
opkg install python3 python3-pip libpython3

May be caused by the upgrade.