Entware/Entware #783: I2Pd update to 2.41 (mipselsf-k3.4)

Regloom, 2022-04-11:
Latest version for mipselsf-k3.4 in repo is 2.38.0-1. It crashes after 10-40 mins of activity ([link to issue](https://github.com/PurpleI2P/i2pd/issues/1738#issuecomment-1069413649)). 

Please update package to the latest version.

There is new version at [PurpleI2P/i2pd](https://github.com/PurpleI2P/i2pd/releases/ )
- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2022-04-15:
Thanks for report.

This package was taken by OpenWrt and it's up to OpenWrt guys for now.
Please, send PR/Issue to them.

closed 2022-04-15 by ryzhovau

r4sas, 2022-04-21:
FYI, OpenWRT has 2.41.0 in master for 2 months.
https://github.com/openwrt/packages/commit/c08bb2df62297967b3bced182bd77fcebac8fc9f

ryzhovau, 2022-04-21:
Looks like we have to stay on 2.38.0 version. Some of our kernel 2.6 based feeds can't handle newer versions.

Regloom, 2022-04-21:
> Looks like we have to stay on 2.38.0 version. Some of our kernel 2.6 based feeds can't handle newer versions.

Ничего не понял, на роутере ядро 3.4
```
~ # uname -r
3.4.113
```

ryzhovau, 2022-04-21:
Набор пакетов единый для всех фидов из http://bin.entware.net/

r4sas, 2022-04-21:
Так а в чем проблема протащить в ваш фид один пакет из мастера?

r4sas, 2022-04-21:
И ядро 2.6 так же без проблем поддерживается (привет из полу-мертвой CentOS 7).

Regloom, 2022-04-22:
Штош собираем сами тогда.

The-BB, 2022-04-22:
Если очень сильно хочется, копайте отсюда - https://github.com/PurpleI2P/i2pd/commit/ed42948051afb308cd7585d4284d286f49829a08 и дальше вверх...

r4sas, 2022-04-22:
Я проверю сборку пакета, посмотрим чего там не так...

r4sas, 2022-04-23:
Дублирую сообщение отправленное в чат в Телеграме:

Пытаюсь собрать для теста пакет из репозитория (см. https://github.com/Entware/Entware/issues/783), но попадаю на ошибку с libffi.

Порядок действий:
```
git clone ...
make package/symlinks
make -j$(nproc) tools/install
make -j$(nproc) toolchain/install
make package/i2pd/compile V=s
```

Чего получаю в ответ:
```
make[2]: Entering directory '/home/wrt/Entware/feeds/packages/libs/libffi'
CFLAGS="<cut>" make -j1 -C /home/wrt/Entware/build_dir/hostpkg/libffi-3.4.2
make[3]: Entering directory '/home/wrt/Entware/build_dir/hostpkg/libffi-3.4.2'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory '/home/wrt/Entware/build_dir/hostpkg/libffi-3.4.2'
make[2]: *** [Makefile:90: /home/wrt/Entware/build_dir/hostpkg/libffi-3.4.2/.built] Error 2
make[2]: Leaving directory '/home/wrt/Entware/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/host-compile#0.31#0.27#0.51
    ERROR: package/feeds/packages/libffi [host] failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/libffi/host/compile] Error 1
make[1]: Leaving directory '/home/wrt/Entware'
make: *** [/home/wrt/Entware/include/toplevel.mk:230: package/i2pd/compile] Error 2
```
ЧЯДНТ?

r4sas, 2022-04-23:
Проблема ясна, для совместимости добавлен код: https://github.com/PurpleI2P/i2pd/commit/1326597226257cbf54eb69f89923c807fc82f76a

Пояснение: для сборки пакетов, как я понимаю, используются собранные на базе musl компиляторы и заголовочные файлы. Из-за этого boost изначально начинает работать с заголовками `netinet` и это не дает нормально инклуднуть заголовки `linux/in6.h`, что и ломает сборку. По этому придется отказаться на системах, где эти заголовки используются, от функционала.

ryzhovau, 2022-04-23:
Класс! Было бы обидно исключить пакет, который когда-то благодаря нам [появился](https://twitter.com/i2porignal/status/574994873159303168).

The-BB, 2022-04-23:
начиная с ядра 3.12 с `#if __UAPI_DEF_*` в `linux/in6.h` можно собирать без "костыля"...

r4sas, 2022-04-24:
> начиная с ядра 3.12 с `#if __UAPI_DEF_*` в `linux/in6.h` можно собирать без "костыля"...

Оно так и будет, когда не будет инклудиться netinet.

Regloom, 2022-04-25:
Отличная работа, спасибо @r4sas @ryzhovau !