Entware/Entware #785: zerotier_1.8.6-1_aarch64-3.10.ipk    NOT working with CoreELEC 19.4 (S905X3)

roisiano, 2022-04-14:
zerotier_1.8.6-1_aarch64-3.10.ipk 
from:
https://bin.entware.net/aarch64-k3.10/

I was using [zerotier_1.8.3-1_aarch64-3.10.ipk](https://bin.entware.net/aarch64-k3.10/archive/zerotier_1.8.3-1_aarch64-3.10.ipk) and, after:
opkg update
opkg upgrade

version zerotier_1.8.6-1 was installed, but NOT working.

zerotier-cli status
-sh: zerotier-cli: not found

I'm using this wit CoreELEC 19.4, in an S905X3 TV-Box (X96 Max+).

I've used Zerotier for 2 years, more or less, without any problem, and upgrded 4-5 times without problem... until now.

After this, I "downgraded" it:
opkg remove zerotier

I downloaded previous version "manually":
https://bin.entware.net/aarch64-k3.10/archive/zerotier_1.8.3-1_aarch64-3.10.ipk

I put it in:
/storage/downloads

and installed it manually:
cd /storage/downloads
opkg install  zerotier_1.8.3-1_aarch64-3.10.ipk

After this, Zerotier working again in this version (1.8.3-1), but NOT with 1.8.6-1

So... it seems a bug with last version.

Thanks for all the packages.

King regards

zyxmon, 2022-04-15:
Thanks for reporting. Fixed via https://github.com/Entware/entware-packages/commit/15c5e9da0d584248897c6570bffef3501bfcdbaa
Please check if it is ok now

roisiano, 2022-04-15:
Yes. New version [zerotier_1.8.6-1a_aarch64-3.10.ipk](https://bin.entware.net/aarch64-k3.10/zerotier_1.8.6-1a_aarch64-3.10.ipk) working OK after upgrading.
Solved.

Thanks

closed 2022-04-15 by ryzhovau