Entware/Entware #786: is not updated

g0blin-NN, 2022-04-17:
**Describe the bug**
on hold https://bin.entware.net/armv7sf-k3.2/Packages.gz

[image]

**Environment (please complete the following information):**
 - Device  Asus RT-AX56U router
 - Firmware version asuswrt-merlin 386.5.2

zyxmon, 2022-04-17:
Probably something wrong with your internet connection. Try another one.
```
/tmp/mnt/WARE/root # opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
/tmp/mnt/WARE/root # opkg upgrade
Upgrading terminfo on root from 6.2-2a to 6.3-1a...
Downloading http://bin.entware.net/armv7sf-k3.2/terminfo_6.3-1a_armv7-3.2.ipk
Upgrading opkg on root from 2021-06-13-1bf042dd-1 to 2022-01-09-2edcfad1-1...
Downloading http://bin.entware.net/armv7sf-k3.2/opkg_2022-01-09-2edcfad1-1_armv7-3.2.ipk
Upgrading libevdev on root from 1.11.0-2 to 1.12.0-1...
Downloading http://bin.entware.net/armv7sf-k3.2/libevdev_1.12.0-1_armv7-3.2.ipk
Upgrading zoneinfo-europe on root from 2021a-1 to 2022a-1...
Downloading http://bin.entware.net/armv7sf-k3.2/zoneinfo-europe_2022a-1_armv7-3.2.ipk
Upgrading diffutils on root from 3.7-3 to 3.8-1...
Downloading http://bin.entware.net/armv7sf-k3.2/diffutils_3.8-1_armv7-3.2.ipk
Removing obsolete file /opt/bin/cmp.
Removing obsolete file /opt/bin/diff.
Upgrading busybox on root from 1.33.1-1 to 1.35.0-2...
Downloading http://bin.entware.net/armv7sf-k3.2/busybox_1.35.0-2_armv7-3.2.ipk
Upgrading glib2 on root from 2.68.1-3 to 2.70.4-2...
Downloading http://bin.entware.net/armv7sf-k3.2/glib2_2.70.4-2_armv7-3.2.ipk
.....
```

g0blin-NN, 2022-04-17:
I only updated via VPN. it seems that Entware's russian IP is no longer available. this is wrong. time will put everything in its place. who is right and who is to blame.

closed 2022-04-17 by g0blin-NN

zyxmon, 2022-04-17:
FYI Russian servers were never used to host Entware.

g0blin-NN, 2022-04-17:
@zyxmon 
Андрей, раньше все работало. сейчас с русского IP обновляюсь. останавливается и висит как на картинке выше. подключаю VPN IP становится из Амстердама. обнова пошла. мне все равно где располагается Entware. самое главное чтобы работало. а если тоже в политику ударились. тут убеждать бесполезно. кто там против кого и за что на самом деле. все расставит время по своим местам. правду то все равно не скрыть. любой пропагандой. только вот возвращаться будет стыдно.

zyxmon, 2022-04-18:
@g0blin-NN - посмотрите закрытые issues - у пользователей разных стран возникают проблемы с обновлением. Некоторые связаны с работой cloudflare - тут никакой политики нет.

zyxmon, 2022-04-18:
Мы сбросили кэш CloudFlare - попробуйте еще раз без VPN (обновить список, установить пакет).

g0blin-NN, 2022-04-18:
@zyxmon 
висит. не скачивается.
```
/tmp/home/root# opkg update
Downloading https://bin.entware.net/armv7sf-k3.2/Packages.gz
^C
```

zyxmon, 2022-04-18:
Почему у Вас https (раньше не заметил)? Вообще говоря, стандартный opkg работает через wget и http.
Можете попробовать скачать по http  файл Packages.gz с помощью разных бинарников wget или curl. Хотя, мне кажется, скачиваться все равно не будет.

g0blin-NN, 2022-04-18:
а я почем знаю, почему https? так сделано в прошивке.
```
C:\Temp>curl https://bin.entware.net/armv7sf-k3.2/Packages.gz -o Packages.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  307k  100  307k    0     0   615k      0 --:--:-- --:--:-- --:--:--  616k
```
```
C:\Temp>wget https://bin.entware.net/armv7sf-k3.2/Packages.gz -O Packages.gz
--2022-04-18 17:15:58--  https://bin.entware.net/armv7sf-k3.2/Packages.gz
Resolving bin.entware.net (bin.entware.net)... 104.21.91.83, 172.67.212.134
Connecting to bin.entware.net (bin.entware.net)|104.21.91.83|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 314784 (307K) [application/x-gzip]
Saving to: 'Packages.gz'

Packages.gz                   100%[=================================================>] 307,41K  --.-KB/s    in 0,1s

2022-04-18 17:15:58 (2,50 MB/s) - 'Packages.gz' saved [314784/314784]
```
а в роутере не хочет.

zyxmon, 2022-04-18:
>> а в роутере не хочет. на ноуте качает и по http и по https. странно.

Так проблема с прошивкой или сертификатами или хз. В чистом Entware проблемы у Вас быть не должно.

g0blin-NN, 2022-04-18:
тогда почему через vpn обновляет?
```
/tmp/home/root# opkg update
Downloading https://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
```

Lichtenshtein, 2022-05-21:
Тоже столкнулся с этой проблемой. И в браузере на пк не качается: ERR_CONNECTION_TIMED_OUT.

https://bin.entware.net/mipselsf-k3.4/Packages.gz