Entware/Entware #788: Rtorrent/Rutorrent webui not working.

fireqpeg, 2022-04-21:
**Describe the bug**
[package name]: A clear and concise description of what the bug is and how to reproduce it. 

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - Device [e.g. Asus RT-AX88U router]
 - Firmware version [e.g. asuswrt-merlin]

fireqpeg, 2022-04-21:
http://bin.entware.net/armv7sf-k3.2

device: RT-AX56U Firmware Version:[386.5_2]

II installed rtorrent, rutorrent. And your webui won't load! I tried to use a different config. Set a different port. It doesn't work either. Please explore and correct this error!

ryzhovau, 2022-04-21:
Look's like we have to rework this package. As a temporary solution, try `ln -sf /opt/bin/php8-fcgi php-fcgi`.

fireqpeg, 2022-04-22:
Thank you help me! (ln -sf php8-fcgi /opt/bin/php-fcgi.) It works like this!

The-BB, 2022-08-31:
fixed via https://github.com/Entware/entware-packages/commit/9e9e8ca83bb0785e777c14abe6826dbb29f80a6f

closed 2022-08-31 by The-BB