Entware/Entware #791: [Request] Navidrone [Request]

jumpsmm7, 2022-04-24:
[image]

# Navidrome Music Server

[image]
[image]
[image]
[image]
[image]
[image]
[image]

## [Check out our Live Demo!](https://www.navidrome.org/demo/)

Navidrome is an open source web-based music collection server and streamer. It gives you freedom to listen to your 
music collection from any browser or mobile device. It's like your personal Spotify!

It is compatible across numerous platforms including LINUX. While their are debian based was to run it, they provide the binaries that would be useful in creating an entware package solution. 

Both Alpine and Arch linux have their own package handlers for this project. It would be nice if it could be ported to Entware repo. 
 
https://github.com/navidrome/navidrome

Precompiled binaries located: https://github.com/navidrome/navidrome/releases/tag/v0.47.5

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

The-BB, 2022-08-25:
please, test - http://bin.entware.net/mipselsf-k3.4/test/navidrome_0.47.5-1_mipsel-3.4.ipk

jumpsmm7, 2022-08-26:
> please, test - http://bin.entware.net/mipselsf-k3.4/test/navidrome_0.47.5-1_mipsel-3.4.ipk

It looks promising I am currently investigating it right now.

spinline, 2023-03-11:
error while loading shared libraries: libtag.so.1: cannot open shared object file: No such file or directory

ryzhovau, 2023-05-23:
As long as discussion hangs, I'm closing this one.
I'll open it again if someone will be interested.

closed 2023-05-23 by ryzhovau