Entware/Entware #801: [Package Request] alist

dante2046, 2022-05-22:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

project URL
https://github.com/Xhofe/alist

this project can mount google drive , router itself as a webdav server 

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

The-BB, 2022-08-25:
please, test - http://bin.entware.net/mipselsf-k3.4/test/alist_2.6.4-1_mipsel-3.4.ipk

The-BB, 2022-09-02:
```
~ # uname -m
aarch64
~ # 
~ # ldd /opt/bin/alist 
	linux-vdso.so.1 (0x0000007f83617000)
	libdl.so.2 => /opt/lib/libdl.so.2 (0x0000007f835d5000)
	libpthread.so.0 => /opt/lib/libpthread.so.0 (0x0000007f835a7000)
	libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x0000007f83582000)
	libc.so.6 => /opt/lib/libc.so.6 (0x0000007f83412000)
	/opt/lib/ld-linux-aarch64.so.1 (0x0000007f835e9000)
~ # 
~ # alist -version
Built At: 2022-08-24 13:32:14 +0000
Go Version: go1.18.4 linux/arm64
Author: Entware
Commit ID: bd52df192
Version: v2.6.4-bd52df192
WebVersion: b095d5d28
~ # 
~ # alist -password
your password: IZRVQpUe
~ #
```
Run `ldconfig`, check "alist" with `ldd`.

dante2046, 2022-09-02:
admin@RT-AX88U-2C90:/tmp/mnt/sda1/entware/bin# uname -m
aarch64
admin@RT-AX88U-2C90:/tmp/mnt/sda1/entware/bin# ldd /opt/bin/alist
        linux-vdso.so.1 (0x0000007f8794c000)
        libdl.so.2 => not found
        libpthread.so.0 => not found
        libgcc_s.so.1 => not found
        libc.so.6 => not found
admin@RT-AX88U-2C90:/tmp/mnt/sda1/entware/bin# alist -version
alist: error while loading shared libraries: libdl.so.2: cannot open shared object file: No such file or directory
admin@RT-AX88U-2C90:/tmp/mnt/sda1/entware/bin# alist -password
alist: error while loading shared libraries: libdl.so.2: cannot open shared object file: No such file or directory

and I had succes run with this [alist-linux-musl-arm64.tar.gz](https://github.com/alist-org/alist/releases/download/v2.6.4/alist-linux-musl-arm64.tar.gz) but  I didn't know haw to  run in background or run in startup

ryzhovau, 2023-05-23:
As long as discussion hangs, I'm closing this one.
I'll open it again if someone will be interested.

closed 2023-05-23 by ryzhovau