Entware/Entware #802: Wiki owncloud appears to be outdated

Jishun, 2022-05-27:
**Describe the bug**
It appears this guide to install owncloud is out dated, the guide uses php7 but the latest package source doesn't provide those anymore 

**Environment (please complete the following information):**
 - Entware feed: [e.g. armelsf-k2.6]
 - Device [e.g. netgear R7000 ]
 - Firmware version [e.g. Fresh tomato]

ryzhovau, 2022-05-27:
Feel free to fix it! You've got all necessary rights to edit it.

sophtron, 2022-05-30:
Well. took a try and finally figured that.. owncloud can't run with php8..

Jishun, 2022-05-30:
Well. took a try and finally figured that.. owncloud can't run with php8.

Jishun, 2022-05-31:
instead, I managed to get nextcloud running with the current available packages, 
I'm unable to get ssl working yet,  because the current lighttpd package appears missing mod_openssl.so
I did it by mostly following the owncloud wiki guide, except installed different packages, 
here is what I had to install 
btw: It took me a long time to fix when I decided not to install it in the path as in the wiki, the problem is, there is a document root setting in php.ini that needs to be updated too, after changing the root in lightttpd.conf 

```
opkg install \
coreutils-stat \
coreutils-mktemp \
git-http \
bzip2 \
php8-cgi \
php8-cli \
lighttpd-mod-fastcgi \
lighttpd-mod-access \
lighttpd-mod-expire \
lighttpd-mod-setenv \
php8-mod-ctype \
php8-mod-curl \
php8-mod-dom \
php8-mod-fileinfo \
php8-mod-gd \
php8-mod-iconv \
php8-mod-mbstring \
php8-mod-pcntl \
php8-mod-pdo \
php8-mod-pdo-sqlite \
php8-mod-session \
php8-mod-simplexml \
php8-mod-sqlite3 \
php8-mod-xml \
php8-mod-xmlwriter \
php8-mod-xmlreader \
php8-mod-openssl \
php8-mod-intl \
php8-mod-zip \
php8-fpm \ 
php8-mod-mysqli \
php8-mod-mbstring \
php8-pecl-redis  \
php8-mod-bcmath \
php8-mod-gmp \
php8-pecl-imagick \
php8-mod-filter \
imagemagick
```

ryzhovau, 2022-05-31:
> I'm unable to get ssl working yet, because the current lighttpd package appears missing mod_openssl.so

Are you sure? 
https://github.com/Entware/Entware/issues/806

Jishun, 2022-05-31:
yeah your are right. thanks, I'm able to install that mod now, I might have been trying with the old ng repo back and forth to mess this up while following the guide with php7

weird, I also came across some topic mentioning the mod_ssl should have been built in to lighttpd(ssl) while it is the `ssl` version. 

anyway, thanks the server is able to start with this installed now, yet not able to serve the content. I'll see what else I can find.

ryzhovau, 2022-07-25:
Closing for now.

closed 2022-07-25 by ryzhovau

muratyuceer, 2023-11-11:
Hi @Jishun

Hi, did you manage to get owncloud working? I have php8 problem like you..