Entware/Entware #805: Tools from shadow-* packages fail to read passwd/shadow/group if it is a symlink

Zulgrib, 2022-05-28:
**Describe the bug**
All shadow-* package fails reading "/opt/etc/{passwd,shadow,group}" if it is a symlink to /etc/{passwd,shadow,group}
For example : 
```
root@DS: /# /opt/sbin/usermod -s /bin/bash root
usermod: cannot open /opt/etc/passwd
```

**Environment (please complete the following information):**
 - Entware feed: x64-k3.2
 - Device : Synology DS918+
 - Firmware version : DSM6

**Expected behavior**
If the file is a symlink, it should try to read the target for the link instead.
If nsswitch.conf is involved, allow setting the file path there or create a compatibility mode (compat2?) where path would be in /etc instead of /etc/opt for specified data sources.

May be related to : 
https://github.com/Entware/Entware/blob/5fcecd7e413c46f44422d982c66480117da005e4/toolchain/glibc/patches/2.27/9999-glibc_etc.patch
https://github.com/Entware/Entware/blob/5fcecd7e413c46f44422d982c66480117da005e4/package/utils/util-linux/patches/600-entware-paths.patch

zyxmon, 2022-05-28:
Shadow-* binaries should be used for alternative Entware installation. In this case files in /opt/etc folder are not symlinks. Use utilities from firmware to manage files in /etc.

closed 2022-05-28 by zyxmon

Zulgrib, 2022-05-29:
firmware may not have tool to change user shell and modifying passwd file by hand isn't safe because humans make mistakes

zyxmon, 2022-05-30:
I prefer alternative Entware installation on my Qnap nases. I think it is preferable for syno. In this case you have diffferent set of users in firmware and in Entware and do not modify system files.