Entware/Entware #813: Entware_1.03 QNAP : The digital signature has expired.

skaox, 2022-06-26:
Hello, everything is in the title.
Thank you in advance.

**Digital signature information**
Name: Silas Mariusz
Signing date; 2020/10/23 13:04:17

**Certificate information**
Issued by: Third party
Issued to: Silas Mariusz
Valid date : 2019/06/27 10:34:10 - 2022/06/26 10:34:10

zyxmon, 2022-06-27:
Please read this topic https://forum.qnap.com/viewtopic.php?f=320&t=166765&p=820567 and other similar topics on Qnap forum.

closed 2022-06-27 by zyxmon

jamieburchell, 2022-06-27:
@zyxmon I'm seeing this too and having read the link you pointed to the solution isn't clear. It's the actual Entware QPKG itself that appears to be reporting the issue.

zyxmon, 2022-06-27:
qnapclub signes packages. for now just enable unsigned packages in  appcenter settings.

skaox, 2022-06-27:
Perfect valid until **2025/04/08** a big **THANK YOU !**

(-: You are a good Samaritan ;-)

[Wiki](https://github.com/Entware/Entware/wiki/Install-on-QNAP-NAS) also need to be updated.