Entware/Entware #819: pip fails with OSError on TomatoUSB/ARM

kcgthb, 2022-07-15:
**Describe the bug**
[python3-pip]: pip fails with OSError on ARM/TomatoUSB

```
# pip3 install speedtest-cli
ERROR: Could not install packages due to an OSError: ("Connection broken: OSError(22, 'Invalid argument')", OSError(22, 'Invalid argument'))
```

Verbose stack:
```
# pip3 -v install speedtest-cli
Using pip 22.0.4 from /opt/lib/python3.10/site-packages/pip (python 3.10)
ERROR: Could not install packages due to an OSError.
Traceback (most recent call last):
  File "/opt/lib/python3.10/site-packages/pip/_vendor/urllib3/response.py", line 438, in _error_catcher
  File "/opt/lib/python3.10/site-packages/pip/_vendor/urllib3/response.py", line 519, in read
  File "/opt/lib/python3.10/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 96, in read
  File "/opt/lib/python3.10/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 74, in _close
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/lib/python3.10/site-packages/pip/_vendor/requests/models.py", line 760, in generate
  File "/opt/lib/python3.10/site-packages/pip/_vendor/urllib3/response.py", line 576, in stream
  File "/opt/lib/python3.10/site-packages/pip/_vendor/urllib3/response.py", line 512, in read
  File "/opt/lib/python3.10/contextlib.py", line 153, in __exit__
  File "/opt/lib/python3.10/site-packages/pip/_vendor/urllib3/response.py", line 455, in _error_catcher
pip._vendor.urllib3.exceptions.ProtocolError: ("Connection broken: OSError(22, 'Invalid argument')", OSError(22, 'Invalid argument'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/lib/python3.10/site-packages/pip/_internal/commands/install.py", line 339, in run
  File "/opt/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 94, in resolve
  File "/opt/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 481, in resolve
  File "/opt/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 348, in resolve
  File "/opt/lib/python3.10/site-packages/pip/_vendor/resolvelib/resolvers.py", line 172, in _add_to_criteria
  File "/opt/lib/python3.10/site-packages/pip/_vendor/resolvelib/structs.py", line 151, in __bool__
  File "/opt/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 155, in __bool__
  File "/opt/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 143, in <genexpr>
  File "/opt/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/found_candidates.py", line 44, in _iter_built
  File "/opt/lib/python3.10/site-packages/pip/_internal/resolution/resolvelib/factory.py", line 294, in iter_index_candidate_infos
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/package_finder.py", line 868, in find_best_candidate
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/package_finder.py", line 809, in find_all_candidates
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/sources.py", line 134, in page_candidates
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/package_finder.py", line 769, in process_project_url
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/collector.py", line 566, in fetch_page
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/collector.py", line 471, in _get_html_page
  File "/opt/lib/python3.10/site-packages/pip/_internal/index/collector.py", line 123, in _get_html_response
  File "/opt/lib/python3.10/site-packages/pip/_vendor/requests/sessions.py", line 542, in get
  File "/opt/lib/python3.10/site-packages/pip/_internal/network/session.py", line 454, in request
  File "/opt/lib/python3.10/site-packages/pip/_vendor/requests/sessions.py", line 529, in request
  File "/opt/lib/python3.10/site-packages/pip/_vendor/requests/sessions.py", line 687, in send
  File "/opt/lib/python3.10/site-packages/pip/_vendor/requests/models.py", line 838, in content
  File "/opt/lib/python3.10/site-packages/pip/_vendor/requests/models.py", line 763, in generate
pip._vendor.requests.exceptions.ChunkedEncodingError: ("Connection broken: OSError(22, 'Invalid argument')", OSError(22, 'Invalid argument'))
```

Installed packages:
```
# opkg list-installed python*
python-pip-conf - 0.1-1
python3 - 3.10.4-1
python3-asyncio - 3.10.4-1
python3-base - 3.10.4-1
python3-cgi - 3.10.4-1
python3-cgitb - 3.10.4-1
python3-codecs - 3.10.4-1
python3-ctypes - 3.10.4-1
python3-dbm - 3.10.4-1
python3-decimal - 3.10.4-1
python3-distutils - 3.10.4-1
python3-email - 3.10.4-1
python3-light - 3.10.4-1
python3-logging - 3.10.4-1
python3-lzma - 3.10.4-1
python3-multiprocessing - 3.10.4-1
python3-ncurses - 3.10.4-1
python3-openssl - 3.10.4-1
python3-pip - 22.0.4-1
python3-pkg-resources - 58.1.0-1
python3-pydoc - 3.10.4-1
python3-readline - 3.10.4-1
python3-setuptools - 58.1.0-1
python3-sqlite3 - 3.10.4-1
python3-unittest - 3.10.4-1
python3-urllib - 3.10.4-1
python3-uuid - 3.10.4-1
python3-xml - 3.10.4-1
```

This is a new issue with Python 3.10, there was no issue with Python 3,9.

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k2.6
 - Device R7000 router
 - Firmware version Tomato Firmware 1.28.0000 -3.5-140 K26ARM USB AIO-64K

zyxmon, 2022-07-17:
No problems on 'Tomato Firmware 1.28.0000 --130_2 K26ARM USB AIO-64K':
```
~ # pip install --upgrade pip setuptools
Requirement already satisfied: pip in /opt/lib/python3.10/site-packages (22.0.4)
Collecting pip
  Downloading pip-22.1.2-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 577.5 kB/s eta 0:00:00
Requirement already satisfied: setuptools in /opt/lib/python3.10/site-packages (58.1.0)
Collecting setuptools
  Downloading setuptools-63.2.0-py3-none-any.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 1.5 MB/s eta 0:00:00
Installing collected packages: setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 58.1.0
    Uninstalling setuptools-58.1.0:
      Successfully uninstalled setuptools-58.1.0
  Attempting uninstall: pip
    Found existing installation: pip 22.0.4
    Uninstalling pip-22.0.4:
      Successfully uninstalled pip-22.0.4
Successfully installed pip-22.1.2 setuptools-63.2.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
~ # pip3 install speedtest-cli
Collecting speedtest-cli
  Downloading speedtest_cli-2.1.3-py2.py3-none-any.whl (23 kB)
Installing collected packages: speedtest-cli
Successfully installed speedtest-cli-2.1.3
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```
Probably your firmware is not compatible with Entware.

closed 2022-07-17 by zyxmon

kcgthb, 2022-07-17:
Unfortunately:
```
# pip install --upgrade pip setuptools                                                           Requirement already satisfied: pip in /opt/lib/python3.10/site-packages (22.0.4)                                        ERROR: Could not install packages due to an OSError: ("Connection broken: OSError(22, 'Invalid argument')", OSError(22, 'Invalid argument')) 
```

There was never any problem on that very same system (same firmware) until the upgrade to Python 3.10. Pip packages could be installed without any issue with Python 3.9 Entware packages.

zyxmon, 2022-07-17:
Try a fresh install. There may be some dependent libraries that opkg failed to upgrade.

kcgthb, 2022-07-17:
Thanks for the suggestion. Unfortunately, a fresh install of Entware (after having formatted the device containing `/opt` and the previous installation) has the exact same issue.

But looking further, the issue seems to be with the `python3-pip (22.0.4-1)` package, because downgrading it to the previous version (`python3-pip (21.2.3-1)`) appears to resolve the problem:

```
# pip install speedtest-cli
ERROR: Could not install packages due to an OSError: ("Connection broken: OSError(22, 'Invalid argument')", OSError(22, 'Invalid argument'))
#
# opkg remove python3-pip
Removing package python3-pip from root...
#
# opkg install http://bin.entware.net/armv7sf-k2.6/archive/python3-pip_21.2.3-1_armv7-2.6.ipk
Downloading http://bin.entware.net/armv7sf-k2.6/archive/python3-pip_21.2.3-1_armv7-2.6.ipk
Installing python3-pip (21.2.3-1) to root...
Configuring python3-pip.
#
# pip install speedtest-cli
Collecting speedtest-cli
Downloading speedtest_cli-2.1.3-py2.py3-none-any.whl (23 kB)
Installing collected packages: speedtest-cli
Successfully installed speedtest-cli-2.1.3
#
```