Entware/Entware #823: [Package Request] git-lfs

Miladiir, 2022-07-29:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.
Loved and hated by developers all around the world: Git Large File Storage.

https://github.com/git-lfs/git-lfs

I think it is fairly well known. It enables ~different~ *better* large file support for git.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

The-BB, 2022-08-25:
please, test - http://bin.entware.net/mipselsf-k3.4/test/git-lfs_3.2.0-1_mipsel-3.4.ipk

The-BB, 2022-08-30:
You can find this package [here](http://bin.entware.net/armv7sf-k2.6/test/) or [here](http://bin.entware.net/armv7sf-k3.2/test/)