Entware/Entware #827: napt66

geyee, 2022-08-03:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

IPv6-IPv6 Network Address Port Translation, or upgrade ip6tables version with nat support.
the project url is https://github.com/mzweilin/napt66.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [x86-64 -kernel 3.4.6] and get a feedback about it.

ryzhovau, 2022-08-03:
What's the killer feature of `napt66` in front of `tayga`?

geyee, 2022-08-04:
> What's the killer feature of `napt66` in front of `tayga`?

Tayga is nat64, while napt66 is ipv6 nat66. IE, napt66 is IPv6 to IPv6 NAT, while taya is IPv6 to IPv4 translation. they are in different direction on packets translation. Napt66 can help  devices behind the router or linux pc (kernel under 3.9.0) connect to IPv6 network if the gateway hava a public IPv6 address, while tayga couldn't.
 As [Linux IPv6 HOWTO (en)](https://mirrors.bieringer.de/Linux+IPv6-HOWTO/nat-netfilter6..html) said, "Since at least Linux kernel version 3.9.0 and using ip6tables since 1.4.18 also Network Address Translation (NAT) is usable". 
Thanks.

ryzhovau, 2023-06-05:
Just figured out that is the kernel module.
To keep Entware firmware independent, we do not apply any kernel specific modules, sorry.

closed 2023-06-05 by ryzhovau