Entware/Entware #830: qbittorrent 4.4.3.1-1 non-latin chars in filenames replaced with dots

JugglerLKR, 2022-08-15:
**Describe the bug**
when downloading torrents with non-latin chars in filenames, those characters written as dots to ext filesystem

**Environment (please complete the following information):**
 - Entware feed: aarch64-k3.10
 - Device RT-AX68U router
 - Firmware version asuswrt-merlin 386.7_2
[image]

zyxmon, 2022-08-15:
cannot confirm it. Tested on aarch64 qnap nas.

closed 2022-08-15 by zyxmon

zyxmon, 2022-08-15:
https://github.com/qbittorrent/qBittorrent/issues/14837

In my case
```
USER=root
SSH_CLIENT=10.0.0.4 1530 2222
SHLVL=1
HOME=/opt/root
TERMINFO=/opt/share/terminfo
SSH_TTY=/dev/pts/1
LOGNAME=root
TEMP=/opt/tmp
TERM=xterm
PATH=/opt/bin:/opt/sbin:/usr/sbin:/usr/bin:/sbin:/bin
LANG=en_US.UTF-8
TMP=/opt/tmp
SHELL=/opt/bin/sh
PWD=/share/CACHEDEV1_DATA/.qpkg/Entware/root
LC_ALL=en_US.UTF-8
SSH_CONNECTION=10.0.0.4 1530 10.0.0.37 2222
```

JugglerLKR, 2022-08-15:
same here
`
TERM=xterm
LANG=en_US.UTF-8
LC_ALL=en_US.UTF-8
`

zyxmon, 2022-08-15:
It looks like qbittorrent bug. I have applied one of the patches and compiled a new version
Try it https://test.entware.net/aarch64-k3.10/qbittorrent_4.4.3.1-2_aarch64-3.10.ipk

PS I have no error report running  qbittorrent-nox manually. Please check if the new version does not have it.

JugglerLKR, 2022-08-15:
> Please check if the new version does not have it.

'Only C and default locale supported with the posix collation implementation' - this error disappeared. but bug with non-latin filenames still exist.

ryzhovau, 2022-08-15:
So it's about interactive vs non-interactive difference. 
I guess no any locale-related variables are provided at startup, so you may add it to start script.

zyxmon, 2022-08-15:
Here is a discussion on the problem - https://githubhelp.com/qbittorrent/qBittorrent/issues/16127

> This is a known problem. According to the bittorrent spec the filenames (and any text string) must be utf8 encoded. Some torrent creating programs are broken and instead of using utf8 they are using the user's locale/charset.
> When libtorrent detects an illegal/invalid unicode point in the string it substitutes it with a dot. (IIRC libtorrent RC_1_1 and earlier substituted it with and underscore).
> 

Please check if Entware locale support is correctly installed:
```
# locale.new -a
C
en_US.utf8
POSIX
ru_RU.utf8
```
Try to add `LC_ALL=en_US.UTF-8` variable to S89qbittorrent script

zyxmon, 2022-08-15:
```
#!/bin/sh

ENABLED=yes
PROCS=qbittorrent-nox
ARGS="--webui-port=9080 --profile=/opt/etc --configuration=entware -d"
PREARGS="LC_ALL=en_US.UTF-8"
DESC=$PROCS
PATH=/opt/sbin:/opt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

. /opt/etc/init.d/rc.func

```

JugglerLKR, 2022-08-16:
> # locale.new -a
> C
> en_US.utf8
> POSIX
> ru_RU.utf8
this is exactly as it is.

> Try to add `LC_ALL=en_US.UTF-8` variable to S89qbittorrent script

tried. didn't help with filenames in qbittorrent, but ruined mc "graphics" and filenames in non-latin chars are in '?' :-D

zyxmon, 2022-08-16:
> but ruined mc "graphics" and filenames in non-latin chars are in '?' :-D
> coz now env list missing LC_ALL= and LANG= altogether.
That is odd. Check the script may be therer is a typo, try another variant.

zyxmon, 2022-08-16:
Here is the script I've suggested - https://test.entware.net/aarch64-k3.10/S89qbittorrent

zyxmon, 2022-08-16:
my typo sorry/
the correct line is 
`export LC_ALL=en_US.UTF-8`

zyxmon, 2022-08-19:
Here is the fixed package - https://test.entware.net/aarch64-k3.10/qbittorrent_4.4.3.1-3_aarch64-3.10.ipk
It has upstream patch and modified startup script. If anybody has similar problem with old torrent files for another architecture - please report = I'll build a test package.

JugglerLKR, 2022-08-31:
> Here is the fixed package

sorry for the slow reply. did opkg -force-reinstall and it works great. thank you very much!

kovacspityu, 2022-11-24:
same error here
 https://bin.entware.net/armv7sf-k3.2/qbittorrent_4.4.3.1-1_armv7-3.2.ipk

JugglerLKR, 2023-07-10:
This issue returned again with qbittorrent - 4.5.3-1 version.