Entware/Entware #831: opkg_download_pkg: Package libx264 is not available from any configured src.

gilgrissom, 2022-08-15:
**Describe the bug**
libffmpeg-full & libx264: Trying to update versions of installed packages, following error is show when upgrading libffmpeg-full

`xxxx@yyyyy:/opt/lib# opkg upgrade libffmpeg-full
Upgrading libffmpeg-full on root from 4.3.2-1 to 5.0.1-1...
Downloading https://bin.entware.net/armv7sf-k2.6/libffmpeg-full_5.0.1-1_armv7-2.6.ipk
Installing libx264 (snapshot-20190324-2245-2) to root...
Collected errors:
 * opkg_download_pkg: Package libx264 is not available from any configured src.
 * opkg_install_pkg: Failed to download libx264. Perhaps you need to run 'opkg update'?`

**Environment (please complete the following information):**
    Entware feed: armv7sf-k3.2
    Device Asus RT-AC3200
    Firmware version Freshtomato 2022.5 K26ARM USB AIO-128K

zyxmon, 2022-08-15:
Your repository is armv7sf-k2.6 (from download URL). It does not have libx264 because of a too old kernel.

closed 2022-08-15 by zyxmon

zyxmon, 2022-08-15:
please correct your info - from download url your repo is armv7sf-k2.6 and from text it is armv7sf-k3.2!

zyxmon, 2022-08-15:
Something wrong with your package index. Run `opkg update`!
```
~ # opkg install libffmpeg-full
Installing libffmpeg-full (5.0.1-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libffmpeg-full_5.0.1-1_armv7-2.6.ipk
Installing alsa-lib (1.2.6.1-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/alsa-lib_1.2.6.1-1_armv7-2.6.ipk
Installing libgmp (6.2.1-1a) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libgmp_6.2.1-1a_armv7-2.6.ipk
Installing libnettle (3.8-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libnettle_3.8-1_armv7-2.6.ipk
Installing libatomic (8.4.0-11) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libatomic_8.4.0-11_armv7-2.6.ipk
Installing libgnutls (3.7.6-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libgnutls_3.7.6-1_armv7-2.6.ipk
Installing libopus (1.3.1-3) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libopus_1.3.1-3_armv7-2.6.ipk
Installing shine (3.1.1-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/shine_3.1.1-1_armv7-2.6.ipk
Installing libx264 (2020-10-26-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libx264_2020-10-26-1_armv7-2.6.ipk
Configuring libopus.
Configuring alsa-lib.
Configuring libatomic.
Configuring libgmp.
Configuring libnettle.
Configuring libgnutls.
Configuring shine.
Configuring libx264.
Configuring libffmpeg-full.

```

gilgrissom, 2022-08-15:
There were indeed index inconsistences. The following fixed it and allowed installation:
```
opkg -V2 --cache /opt/tmp/ocache install --nodeps libx264_snapshot-20190324-2245-2_armv7-2.6.ipk
opkg -V2 --cache /opt/tmp/ocache install --nodeps libx264_2020-10-26-1_armv7-2.6.ipk
opkg remove --nodeps --force-depends --force-removal-of-dependent-packages libx264
opkg -V2  install libx264_2020-10-26-1_armv7-2.6.ipk
```

**libx264**
```
Now clean
Package: libx264
Version: 2020-10-26-1
Depends: libc, libssp, librt, libpthread, libpthread
Status: install user installed
Section: libs
Architecture: armv7-2.6
Size: 842487
Filename: libx264_2020-10-26-1_armv7-2.6.ipk
Description: x264 is a free software library for encoding
 video streams into the H.264/MPEG-4 AVC compression format.
Installed-Time: 1660598268
```
**libffmpeg-full**
```
Package: libffmpeg-full
Version: 5.0.1-1
Depends: libc, libssp, librt, libpthread, libpthread, libpthread, zlib, libbz2, alsa-lib, libgnutls, libopus, shine, libx264
Provides: libffmpeg
Status: install user installed
Section: libs
Architecture: armv7-2.6
Size: 6387509
Filename: libffmpeg-full_5.0.1-1_armv7-2.6.ipk
Description: FFmpeg is a a software package that can record, convert and stream digital
 audio and video in numerous formats.

 FFmpeg licensing / patent issues are complex. It is the reponsibility of the
 user to understand any requirements in this regard with its usage. See:
 https://ffmpeg.org/legal.html for further information.
 .
 This package contains full-featured FFmpeg shared libraries.
Installed-Time: 1660596457
```