Entware/Entware #833: python3-qrcode missing from https://bin.entware.net/armv7sf-k2.6/ only available in /armv7sf-k2.6/archive/

gilgrissom, 2022-08-16:
Describe the bug
python3-qrcode
`python3-qrcode_7.3.1-1_armv7-2.6.ipk`
missing from:
`https://bin.entware.net/armv7sf-k2.6/`
only available in:
`https://bin.entware.net/armv7sf-k2.6/archive/`

Environment:
Entware feed: armv7sf-k2.6
Device Asus RT-AC3200
Firmware version Freshtomato 2022.5 K26ARM USB AIO-128K

The-BB, 2022-08-16:
not a bug, see details https://github.com/openwrt/packages/commit/26f99270d276033569fd4e7ec079f7947b75bc17

gilgrissom, 2022-08-16:
Thanks for the heads-up, however I'm not sure I fully grab the meaning, here:

```
python-packages: Remove unused library packages

With the removal of Seafile, these library packages no longer have any
in-repo users. They will be imported into the abandoned packages
repo[1].
```
Or how python3-qrcode relates to Seafile:
```
xxxx@yyyy:/opt/tmp# opkg info python3-qrcode
Package: python3-qrcode
Version: 7.3.1-1
Depends: libc, libssp, librt, libpthread, python3, python3-setuptools, python3-pillow
Status: install user installed
Architecture: armv7-2.6
Installed-Time: 1660617943
```

closed 2023-05-23 by ryzhovau