Entware/Entware #834: [Package Request] Coturn

Tokarak, 2022-08-16:
**Describe the solution you'd like**
[See](coturn/coturn) for project. Coturn is an opensource TURN server. Entware does not yet have a turn server.

I am aware that turn severs in production environments are not to be run on embedded devices, but there is a perfectly good use case to utilise turn servers for personal use, in which case they do not use many resources. I need coturn for direct webrtc calls.

Official build instructions can be found [here](coturn/coturn/blob/master/rpm/build.instructions.txt)
- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform (Asuswrt-Merlin) and provide feedback about it.

The-BB, 2022-08-25:
please, test - http://bin.entware.net/mipselsf-k3.4/test/coturn_4.5.2-4_mipsel-3.4.ipk

Tokarak, 2022-09-01:
It runs perfectly (aarch64), as far as I can tell right now. I had to change proc-user in the default config, but thats not a problem. Further testing may be required, and I'll update this thread if I find anything. Thank you very much.

closed 2022-12-01 by The-BB