Entware/Entware #835: Byobu

blucey, 2022-08-19:
Please add a package for byobu!

Byobu is a wrapper around tmux or screen that makes the interaction with those tools much easier, as well as adds additional features.

It can be found at:
https://launchpad.net/byobu
and:
https://github.com/dustinkirkland/byobu

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

The-BB, 2022-08-25:
please, test - http://bin.entware.net/x64-k3.2/test/byobu_5.133-1_all.ipk and http://bin.entware.net/x64-k3.2/test/byobu-utils_5.133-1_all.ipk (deps: http://bin.entware.net/x64-k3.2/test/libnewt_0.52.21-4_x64-3.2.ipk http://bin.entware.net/x64-k3.2/test/python3-newt_0.52.21-4_x64-3.2.ipk)

closed 2022-12-01 by The-BB

blucey, 2023-01-17:
My apologies for missing this.  It looks to work perfectly.  It would be great if it could be added to the package list!