Entware/Entware #837: shntool patches (from debian project)

bayonne, 2022-08-27:
The version of shntool supplied with entware cannot handle 24-bit flac files properly (on my Qnap NAS intel x86_64).

For example, I get this in response to ``shntool hash``:

`shntool [hash]: warning: unsupported format 0xfffe (Unknown) while processing file: [01. Walk On.flac]
`

A bug report mentions the issue: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=950803

It turns out that a patch addresses this issue. It can be found at https://salsa.debian.org/debian/shntool/-/tree/master/debian/patches along with a few other patches.

zyxmon, 2022-08-28:
Please check if this package fixes the problem - https://test.entware.net/x64-k3.2/shntool_3.0.10-3_x64-3.2.ipk

`opkg install https://test.entware.net/x64-k3.2/shntool_3.0.10-3_x64-3.2.ipk`

bayonne, 2022-08-28:
Hello.,
I tried to shntool hash 24-bit flac files with this test package.and it worked.

zyxmon, 2022-08-28:
fixed via https://github.com/Entware/rtndev/commit/8b8e2e4e6eddb72bb3a817c295beef82159c4474

closed 2022-08-28 by zyxmon