Entware/Entware #841: dropbear version needs updating to fix a bug

tsiang888, 2022-09-09:
**Describe the bug**
dropbear does not invoke sftp server - the version in entware is Dropbear server v2018.76 and the latest is v2022.XX.   The problem is described at https://svn.dd-wrt.com/ticket/6393 

"Looking at the dropbear's default_options.h file in current dd-wrt sources [​https://svn.dd-wrt.com/browser/src/router/dropbear/default_options.h#L294](https://svn.dd-wrt.com/browser/src/router/dropbear/default_options.h#L294), it seems like although the SFTPSERVER_PATH is well defined it misses the DROPBEAR_SFTPSERVER define as seen in current dropbear sources [​https://secure.ucc.asn.au/hg/dropbear/file/default/default_options.h#l249].
This leads to dropbear not executing the sftp-server even if sftp-server binary is present in /usr/libexec/"

The workaround listed in https://svn.dd-wrt.com/ticket/6393 does work e.g. sftp -s /opt/libexec/sftp-server root@192.168.0.1 however I don't have the ability to specify command line parameters in the backup program I would like to use (Arq).

**Environment (please complete the following information):**
 - Entware feed: http://bin.entware.net/armv7sf-k3.2/installer/generic.sh 
 - Device Netgear R7000
 - Firmware version  DD-WRT v3.0-r37305 std (10/10/18)

zyxmon, 2022-09-10:
```
# cat /opt/etc/entware_release
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=8.4.0
gcc_flags=-O2 -pipe -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.27
/share/CACHEDEV1_DATA/.qpkg/Entware/root # opkg list-installed | grep dropbear
dropbear - 2022.82-1a

```

closed 2022-09-10 by zyxmon

tsiang888, 2022-09-10:
So I don't know where you get the dropbear source from - but if you I execute dropbear -help from the dd-wrt command line I get this:

root@Maui:~# dropbear -help
Dropbear server v2018.76 https://matt.ucc.asn.au/dropbear/dropbear.html
Usage: dropbear [options]
-b bannerfile   Display the contents of bannerfile before user login
                (default: none)
-r keyfile  Specify hostkeys (repeatable)
                defaults: 
                rsa /tmp/root/.ssh/ssh_host_rsa_key
-R              Create hostkeys as required
-F              Don't fork into background
-E              Log to stderr rather than syslog
-m              Don't display the motd on login
-w              Disallow root logins
-G              Restrict logins to members of specified group
-s              Disable password logins

tsiang888, 2022-09-10:
So I don't know where you get the dropbear source from - but if you I execute dropbear -help from the dd-wrt command line I get this:

***@***.***:~# dropbear -help
Dropbear server v2018.76 https://matt.ucc.asn.au/dropbear/dropbear.html <https://matt.ucc.asn.au/dropbear/dropbear.html>
Usage: dropbear [options]
-b bannerfile Display the contents of bannerfile before user login
(default: none)
-r keyfile Specify hostkeys (repeatable)
defaults:
rsa /tmp/root/.ssh/ssh_host_rsa_key
-R Create hostkeys as required
-F Don't fork into background
-E Log to stderr rather than syslog
-m Don't display the motd on login
-w Disallow root logins
-G Restrict logins to members of specified group
-s Disable password logins

zyxmon, 2022-09-10:
https://bin.entware.net/armv7sf-k3.2/Packages.html

Just update your packages!

tsiang888, 2022-09-10:
Package is updated - the problem seems to be the package contains an old version of dropbear that does not allow the sftp subsystem to execute.

***@***.***:/opt# opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
***@***.***:/opt# opkg upgrade
***@***.***:/opt# opkg list dropbear
dropbear - 2022.82-1a - A small SSH2 server/client designed for small memory environments.
***@***.***:/opt# opkg install dropbear
Package dropbear (2022.82-1a) installed in root is up to date.
***@***.***:/opt# dropbear --help
Invalid option --
Dropbear server v2018.76 https://matt.ucc.asn.au/dropbear/dropbear.html
Usage: dropbear [options]
-b bannerfile   Display the contents of bannerfile before user login
                (default: none)
-r keyfile  Specify hostkeys (repeatable)
                defaults: 
                rsa /tmp/root/.ssh/ssh_host_rsa_key
-R              Create hostkeys as required
-F              Don't fork into background
-E              Log to stderr rather than syslog
-m              Don't display the motd on login
-w              Disallow root logins
-G              Restrict logins to members of specified group
-s              Disable password logins
-g              Disable password logins for root
-B              Allow blank password logins
-T              Maximum authentication tries (default 3)
-j              Disable local port forwarding
-k              Disable remote port forwarding
-a              Allow connections to forwarded ports from any host
-c command      Force executed command
-p [address:]port
                Listen on specified tcp port (and optionally address),
                up to 10 can be specified
                (default port is 22 if none specified)
-P PidFile      Create pid file PidFile
                (default /var/run/dropbear.pid)
-W <receive_window_buffer> (default 24576, larger may be faster, max 1MB)
-K <keepalive>  (0 is never, default 0, in seconds)
-I <idle_timeout>  (0 is never, default 0, in seconds)
-V    Version
***@***.***:/opt#

zyxmon, 2022-09-11:
Have you installed `opkg install openssh-sftp-server`? sftp-server binary is located in `/opt/libexec` folder. Not in /usr!

tsiang888, 2022-09-11:
Hi,

Yes it is installed:

***@***.***:~# opkg install openssh-sftp-server
Package openssh-sftp-server (9.0p1-1) installed in root is up to date.
***@***.***:~# ls -l /opt/libexec
lrwxrwxrwx    1 root     root            18 Sep  8 18:06 sftp-server -> ../lib/sftp-server
***@***.***:~# 

I can invoke sftp if I manually specify the subsystem:

tsiang-air:bin tsiang$ sftp -s /opt/libexec/sftp-server ***@***.***
DD-WRT v3.0-r37305 std (c) 2018 NewMedia-NET GmbH
Release: 10/10/18

Connected to 192.168.0.1.
sftp> 

Unfortunately I can’t specify any command line parameters in the backup program I am using (Arq).

Thanks,

DaveT

zyxmon, 2022-09-11:
I have checked sftp connection on different platforms including armv7sf-k3.2 and it works. There are some problems on nonstandard port. Use 22 standard port - and it should be OK.

tsiang888, 2022-09-11:
Thanks for all your help Andrey,

You’re right the package is fine.  I figured out the problem was that my dd-wrt image already had ssh support via dropbear  (but not sftp).

I had enabled sshd on the dd-wrt image so it was invoking it’s old faulty version of dropbear and I guess this prevented the entware version from being executed.

After disabling sshd in the dd-wrt image via GUI and rebooting the entware version is now executing and I am able to sftp successfully.

Thanks again!

DaveT