Entware/Entware #843: Installation issue on XT8

ricardoaveloso, 2022-09-12:
When installing, /opt exists but folders inside /opt exist but are symbolic links. 
Note that the installation manual for Asus stock firmware only warns about /opt being a symbolic link, but not about the folders within it, so stated workaround doesn't work.

Error: 

`admin@ZenWiFi_XT8-EA10:/tmp/home/root# wget -O - http://bin.entware.net/armv7sf-k3.2/installer/generic.sh | sh
--2022-09-12 17:35:35--  http://bin.entware.net/armv7sf-k3.2/installer/generic.sh
Resolving bin.entware.net... 188.114.97.0, 188.114.96.0, 2a06:98c1:3120::, ...
Connecting to bin.entware.net|188.114.97.0|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2765 (2.7K) [application/octet-stream]
Saving to: 'STDOUT'

-                                                                                                          100%[=======================================================================================================================================================================================================================================================================================>]   2.70K  --.-KB/s    in 0s

2022-09-12 17:35:35 (76.6 MB/s) - written to stdout [2765/2765]

Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
mkdir: can't create directory '/opt/bin': No such file or directory
mkdir: can't create directory '/opt/etc': No such file or directory
mkdir: can't create directory '/opt/lib/': No such file or directory
mkdir: can't create directory '/opt/tmp': No such file or directory
mkdir: can't create directory '/opt/var/': Read-only file system
Info: Opkg package manager deployment...
/opt/bin/opkg: No such file or directory
chmod: /opt/bin/opkg: No such file or directory
/opt/etc/opkg.conf: No such file or directory
Info: Basic packages installation...
sh: /opt/bin/opkg: not found
sh: /opt/bin/opkg: not found
chmod: /opt/tmp: No such file or directory
ln: /opt/etc/passwd: No such file or directory
ln: /opt/etc/group: No such file or directory
ln: /opt/etc/shadow: No such file or directory
ln: /opt/etc/gshadow: No such file or directory
ln: /opt/etc/localtime: No such file or directory
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to $PATH variable
Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
admin@ZenWiFi_XT8-EA10:/tmp/home/root#
`

**Environment:**
 - Entware feed: armv7sf-k3.2
 - Device: Asus XT8
 - Firmware version: stock asuswrt (3.0.0.4.386_49873-g72159cc)

ryzhovau, 2022-09-12:
We not ready to provide personal support, sorry. 
You may find more information on [SNB forum](https://www.snbforums.com/forums/asus-wireless.37/), most of advanced Asus users hangs there, AFAIK.

closed 2022-09-12 by ryzhovau

zyxmon, 2022-09-12:
As a possible solution - read here (Russian) https://forums.zyxmon.org/viewtopic.php?f=5&t=5797
This is on installing Emtware in chroot environment.

ricardoaveloso, 2022-09-13:
> We not ready to provide personal support, sorry. You may find more information on [SNB forum](https://www.snbforums.com/forums/asus-wireless.37/), most of advanced Asus users hangs there, AFAIK.

Sorry, I thought the install guide was supported. 
OK, I understand now it is not. 
Appreciate you taking the time to clear that out. 

As a heads up, the installation process doesn't work (at least not for the XT8), perhaps you might want to at least warn people about it.
If by any chance you want to explain to people why: 

- running the process without the workaround fails to set the nvram variable correctly: after reboot only the first line is kept in each of the two variables (before reboot all lines are there)
- using the workaround also fails (/opt exists)

zyxmon, 2022-09-13:
Any registered GitHub user can edit wiki - you can warn XT8 users yourself.