Entware/Entware #846: Error when compiling nethack: error: macro "__has_attribute" requires an identifier

sryze, 2022-09-25:
**Describe the bug**
[nethack]: Error when compiling package nethack-3.6.6

```
make[3]: Entering directory '/home/entware/Entware/feeds/rtndev/nethack'
touch /home/entware/Entware/build_dir/hostpkg/nethack-3.6.6/src/{config,hack}.h-t
make -C /home/entware/Entware/build_dir/hostpkg/nethack-3.6.6/util makedefs CFLAGS="-I/home/entware/Entware/staging_dir/host/include -I/home/entware/Entware/staging_dir/hostpkg/include -I/home/entware/Entware/staging_dir/target-mips_mips32r2_glibc-2.27/host/include -I../include"
make[4]: Entering directory '/home/entware/Entware/build_dir/hostpkg/nethack-3.6.6/util'
cc -I/home/entware/Entware/staging_dir/host/include -I/home/entware/Entware/staging_dir/hostpkg/include -I/home/entware/Entware/staging_dir/target-mips_mips32r2_glibc-2.27/host/include -I../include   -c -o makedefs.o makedefs.c
In file included from /usr/include/features.h:486,
                 from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from ../include/global.h:9,
                 from ../include/config.h:581,
                 from makedefs.c:10:
/usr/include/x86_64-linux-gnu/sys/cdefs.h:405:73: error: macro "__has_attribute" requires an identifier
  405 | #if __GNUC_PREREQ (3,4) || __glibc_has_attribute (__warn_unused_result__)
      |                                                                         ^
make[4]: *** [<builtin>: makedefs.o] Error 1
make[4]: Leaving directory '/home/entware/Entware/build_dir/hostpkg/nethack-3.6.6/util'
make[3]: *** [Makefile:130: /home/entware/Entware/build_dir/hostpkg/nethack-3.6.6/.built] Error 2
make[3]: Leaving directory '/home/entware/Entware/feeds/rtndev/nethack'
time: package/feeds/rtndev/nethack/host-compile#0.42#0.21#0.65
    ERROR: package/feeds/rtndev/nethack [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/rtndev/nethack/host/compile] Error 1
make[2]: Leaving directory '/home/entware/Entware'
make[1]: *** [package/Makefile:110: /home/entware/Entware/staging_dir/target-mips_mips32r2_glibc-2.27/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/entware/Entware'
make: *** [/home/entware/Entware/include/toplevel.mk:230: world] Error 2
```

**Environment (please complete the following information):**
 - Entware feed: [mips-3.4]
 - Device [not relevant]
 - Firmware version [not relevant]

Build host: Ubuntu 22.04

lurch, 2023-02-17:
I also just ran into this whilst building on Ubuntu 22.04 - I temporarily worked around it by running `make menuconfig` and disabling the option to build nethack :wink: 

Looks like [NetHack 3.6.7](https://www.nethack.org/v367/release.html) includes a fix for this though:
> code in include/tradstdc.h was trying to suppress warn_unused result by defining warn_unused_result to an empty string and that began causing a build error within a system-supplied header file cdefs.h when using Ubuntu Impish 21.10; disable that for any Linux and macOS gcc unless GCC_URWARN is defined to force it back into effect 

so https://github.com/Entware/rtndev/tree/master/nethack needs to update to the latest version?

ryzhovau, 2023-02-17:
Works on Debian stable. You may try [Docker image](https://github.com/Entware/docker) for reproducible environment.
We'll bump nethack version soon.

lurch, 2023-02-17:
> You may try [Docker image](https://github.com/Entware/docker) for reproducible environment.

Ah nice - might be worth mentioning that in https://github.com/Entware/Entware/wiki/Compile-packages-from-sources ?

closed 2023-05-23 by ryzhovau