Entware/Entware #848: Package update request: dnscrypt-proxy2

ADV-R, 2022-09-26:
Release 2.1.2:

Support for DoH over HTTP/3 (DoH3, HTTP over QUIC) has been added. Compatible servers will automatically use it. Note that QUIC uses UDP (usually over port 443, like DNSCrypt) instead of TCP.

In previous versions, memory usage kept growing due to channels not being properly closed, causing goroutines to pile up. This was fixed, resulting in an important reduction of memory usage. Thanks to @lifenjoiner for investigating and fixing this!

DNS64: CNAME records are now translated like other responses. Thanks to @ignoramous for this!

A relay whose name has been configured, but doesn't exist in the list of available relays is now a hard error. Thanks to @lifenjoiner!

Mutexes/locking: bug fixes and improvements, by @ignoramous - Official packages now include linux/riscv64 builds.

dnscrypt-proxy -resolve now reports if ECS (EDNS-clientsubnet) is supported by the server.

dnscrypt-proxy -list now includes ODoH (Oblivious DoH) servers.

Local DoH: queries made using the GET method are now handled.

The service can now be installed on OpenRC-based systems.

PTR queries are now supported for cloaked domains. Contributed by Ian Bashford, thanks!

jberlet, 2022-11-01:
Any chance of getting this updated in entware? Thank you!

closed 2023-01-20 by ADV-R

ryzhovau, 2023-01-21:
Oh, sorry. Looks like we forgot to inform you about last massive update.

Lichtenshtein, 2023-02-10:
Please update to 2.1.4:

- Fixes a regression from version 2.1.3: when cloaking was enabled, blocked responses were returned for records that were not A/AAAA/PTR even for names not in the cloaked list