Entware/Entware #857: [Package Request] newset build Igmpproxy

AfromD, 2022-10-29:
The release version of igmpproxy 0.3 is discarding broadcasts to the router which I need forwarded to my Sonos players in another subnet.
After release, [Issue #54 was fixed ](https://github.com/pali/igmpproxy/commit/cfad33001fc5031399bdd073048ca86889378678)to change over from hardcoded blacklicting to explicit blacklisting. This commit no longer automtically discards the packets Sonos is sending out for device discovery.
Although the fix is more than a year old now, there is no new release version, so this fix is not in the entware repository version. I would appreciate to include a current build with this fix in the repository.

AfromD, 2022-10-29:
Release version 0.4 of igmpproxy has now been released. I would very much appreciate an update of the repo.
Thanks!

AfromD, 2022-12-28:
I see the newest build is not included in the repo update,
I am willing to test 0.4. 
My platform is armv7sf-k3.2

ryzhovau, 2023-05-23:
This package came from OpenWrt upstream with no any changes. So we'll keep the same version as in upstream to keep syncing process as simple as possible.

closed 2023-05-23 by ryzhovau