Entware/Entware #858: iotop error due to python3 error

Lem970, 2022-10-29:
**Describe the bug**
iotop: I'm sorry to bother you, Please forgive me if I'm not in the right place.

I just installed iotop with its dependencies (python 3.10 among them), but it's impossible to succesfully launch it:

```
# iotop
Traceback (most recent call last):
  File "/opt/sbin/iotop", line 17, in <module>    
main()
  File "/opt/lib/python3.10/site-packages/iotop/ui.py", line 746, in main
    main_loop()
  File "/opt/lib/python3.10/site-packages/iotop/ui.py", line 736, in <lambda>
    main_loop = lambda: run_iotop(options)
  File "/opt/lib/python3.10/site-packages/iotop/ui.py", line 620, in run_iotop
    return curses.wrapper(run_iotop_window, options)
  File "/opt/lib/python3.10/curses/__init__.py", line 94, in wrapper
  File "/opt/lib/python3.10/site-packages/iotop/ui.py", line 610, in run_iotop_window
    process_list = ProcessList(taskstats_connection, options)
  File "/opt/lib/python3.10/site-packages/iotop/data.py", line 397, in __init__
    self.update_process_counts()
  File "/opt/lib/python3.10/site-packages/iotop/data.py", line 463, in update_process_counts
    return (total_read, total_write), self.vmstat.delta()
  File "/opt/lib/python3.10/site-packages/iotop/vmstat.py", line 46, in delta
    delta = now[0] - self.vmstat[0], now[1] - self.vmstat[1]
TypeError: unsupported operand type(s) for -: 'str' and 'str'
```

I don't know anything about python, but the problem looks related to wrong types of variables, doesn't it?
This reminds me of some changes between python2 and python3 definitions, maybe.

I've already removed any other python  version from the system, but the one provided by opkg. 
Can I do something?
Thank you anyway, for any reply and for your time.

Bye. 

**Environment (please complete the following information):**

- Entware feed: aarch64-k3.10
 
- Device: Asustor NAS 1102T

- Firmware version: ADM version: 4.1.0RLQ1   BIOS version: 1.14

zyxmon, 2022-10-29:
Just use iotop from repo - no problems on qnap aarch64 nas
```
 # opkg install iotop
Installing iotop (0.6-20200108-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/iotop_0.6-20200108-1_aarch64-3.10.ipk
Installing libpython3 (3.10.5-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libpython3_3.10.5-1_aarch64-3.10.ipk
Installing python3-base (3.10.5-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/python3-base_3.10.5-1_aarch64-3.10.ipk
Installing python3-light (3.10.5-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/python3-light_3.10.5-1_aarch64-3.10.ipk
Installing python3-ncurses (3.10.5-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/python3-ncurses_3.10.5-1_aarch64-3.10.ipk
Installing python3-ctypes (3.10.5-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/python3-ctypes_3.10.5-1_aarch64-3.10.ipk
Configuring libpython3.
Configuring python3-base.
Configuring python3-light.
Configuring python3-ncurses.
Configuring python3-ctypes.
Configuring iotop.
 # iotop
```

closed 2022-10-29 by zyxmon

Lem970, 2022-10-29:
Found.
On my system, in /proc/vmstat there's no line beginning with "pgpgin" nor with "pgpgout". So the python code in /opt/lib/python3.10/site-packages/iotop/vmstat.py gives a TypeError.

Workaround: in /opt/lib/python3.10/site-packages/iotop/vmstat.py change
`pgpgin = ''` to `pgpgin = 0`
and 
`pgpgout = ''` to `pgpgout = 0`

Bye.