Entware/Entware #863: [Package Request]

Quid33, 2022-11-05:
Hello

I suggest "Home Assistant" : Open source home automation 

URL : https://www.home-assistant.io/

A good solution dor home automation.
Based in python3.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2022-11-05:
Please, provide a feedback.

http://bin.entware.net/mipselsf-k3.4/test/HA/

Quid33, 2022-11-05:
Thnks but this Package don't work on Arm V7 :-(

steph@Syno1:~/perso$ sudo opkg install homeassistant_2022.10.5-1_mipsel-3.4.ipk
Unknown package 'homeassistant'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for homeassistant found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package homeassistant.

steph@Syno1:~/perso$ cat /proc/cpuinfo
Processor       : ARMv7 Processor rev 1 (v7l)
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x2
CPU part        : 0xc09
CPU revision    : 1
Hardware        : Comcerto 2000 EVM

ryzhovau, 2022-11-05:
Compiled just like you asked: mipselsf-k3.4 :(
I'm still don't know what Entware feed you are using, so I leave this issue open for now.

Quid33, 2022-11-05:
Sorry, I havent modified the platform 

On my Synology it's [armv7sf-k2.6/] or [armv7sf-k3.2/]

ryzhovau, 2023-06-02:
So, test files proven as half-working.
Also, many changes to upstream (dependent) packages needed. https://github.com/Entware/Entware/issues/850

Won't do that, sorry.