Entware/Entware #864: Possible to add x265 and webp for ffmpeg

fuji246, 2022-11-08:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

hevc and webp are very popular and efficient codec widely used nowadays. Is that possible to enable those for ffmpeg.

I've tried to do this but having some problem with webp so far. I've added "--enable-libwebp" and "+libwebp +libwebpmux",  it compiles successfully, however when I try to run it on mipsel, it gives the following error:

```
root@OpenWrt:~# /opt/bin/ffmpeg -i /mnt/sdb1/jeromy/Photos/master/2022/11/05/20221105_42926.mp4 -y -an -vframes 1 -vf scale=640:-2 20110405_42940_75_0.webp
ffmpeg version 5.0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 8.4.0 (OpenWrt GCC 8.4.0 r2019-c1ce3339)
  configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --enable-libwebp --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  WARNING: library configuration mismatch
  avutil      configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  avcodec     configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  avformat    configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  avdevice    configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  avfilter    configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  swscale     configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  swresample  configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  postproc    configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
  libavutil      57. 17.100 / 57. 17.100
  libavcodec     59. 18.100 / 59. 18.100
  libavformat    59. 16.100 / 59. 16.100
  libavdevice    59.  4.100 / 59.  4.100
  libavfilter     8. 24.100 /  8. 24.100
  libswscale      6.  4.100 /  6.  4.100
  libswresample   4.  3.100 /  4.  3.100
  libpostproc    56.  3.100 / 56.  3.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/mnt/sdb1/jeromy/Photos/master/2022/11/05/20221105_42926.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    comment         : vid:v0d00fg10000cdifnhbc77u6sb0lj6pg
    information     : {"com.bytedance.info": "{}"}
    encoder         : Lavf58.45.100
  Duration: 00:00:26.98, start: 0.000000, bitrate: 1209 kb/s
  Stream #0:0[0x1](und): Video: h264 (avc1 / 0x31637661), yuv420p(tv, smpte170m/bt470bg/smpte170m, progressive), 720x1280 [SAR 1:1 DAR 9:16], 1075 kb/s, 30 fps, 30 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
  Stream #0:1[0x2](und): Audio: aac (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
      vendor_id       : [0][0][0][0]
Automatic encoder selection failed for output stream #0:0. Default encoder for format webp (codec webp) is probably disabled. Please choose an encoder manually.
Error selecting an encoder for stream 0:0
```

list the codec, seems encoder is not available:

```
root@OpenWrt:~# ffmpeg -codecs | grep webp
ffmpeg version 5.0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 8.4.0 (OpenWrt GCC 8.4.0 r2019-c1ce3339)
  configuration: --enable-cross-compile --cross-prefix=mipsel-openwrt-linux-gnu- --arch=mipsel --cpu=mips32r2 --target-os=linux --prefix=/opt --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --enable-libwebp --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-fast-unaligned --disable-runtime-cpudetect --disable-x86asm --enable-gnutls --enable-libopus --enable-small --enable-libshine --enable-gpl --enable-libx264
[... cut ...]
I don't spot any issues with the compile output, but the above "WARNING: library configuration mismatch" looks very confusing to me, and I've no idea why above "configuration" for avcodec has no webp there.

Any thoughts?

Thanks!

- [ x] No similar package already added and/or this one got some new killer feature(s),
- [ x] Package is in active development (at least, not abandoned for 5 years or more),
- [x ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

fuji246, 2022-11-09:
I've checked x86_64, same build parameter, libwebp is linked

```
ldd build_dir/target-x86_64_glibc-2.27/ffmpeg-full/ffmpeg-5.0.1/libavcodec/libavcodec.so
	linux-vdso.so.1 (0x00007ffd5175f000)
	libswresample.so.4 => not found
	libavutil.so.57 => not found
	libwebpmux.so.3 => /usr/lib/x86_64-linux-gnu/libwebpmux.so.3 (0x00007f92308cd000)
	libwebp.so.7 => not found
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f923077e000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f9230762000)
	libmp3lame.so.0 => /usr/lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f92306e8000)
	libopus.so.0 => /usr/lib/x86_64-linux-gnu/libopus.so.0 (0x00007f9230689000)
	libx264.so.161 => not found
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9230666000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9230474000)
	libwebp.so.6 => /usr/lib/x86_64-linux-gnu/libwebp.so.6 (0x00007f923040a000)
```

ryzhovau, 2023-05-23:
We trying to keep balance between available features on existing Entware targets, as long as keeping upstream sync as simple as possible. This config came from OpenWrt with no changes, no plans to change it, sorry.

closed 2023-05-23 by ryzhovau