Entware/Entware #866: Cannot install on Asus RT-AX58U

kingozone, 2022-11-16:
**Issue**
Cannot install on stock Asus firmware. Continue getting this log output below after following instructions on [https://github.com/Entware/Entware/wiki/Install-on-Asus-stock-firmware](url). Is this because my arm version is v7l whereas the install guide lists the install link as having armv7sf?

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k3.2
 - Device: Asus RT-AX58U v2 (AX3000)
 - Firmware version: Stock Asus Firmware - 4.19.183 armv7l

**_Note: I was not able to chmod 644 .wget-hsts intially until having create the file by executing "touch .wget-hsts"_**

Terminal Output:

ozone@RT-AX58U-8530:/tmp/home/root# uname -rm
4.19.183 armv7l
ozone@RT-AX58U-8530:/tmp/home/root# wget -O - http://bin.entware.net/armv7sf-k3.
2/installer/generic.sh | sh
--2022-11-15 19:27:44--  http://bin.entware.net/armv7sf-k3.2/installer/generic.sh
Resolving bin.entware.net... 172.67.212.134, 104.21.91.83, 2606:4700:3032::ac43:d486, ...
Connecting to bin.entware.net|172.67.212.134|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2765 (2.7K) [application/octet-stream]
Saving to: 'STDOUT'

-                   100%[===================>]   2.70K  --.-KB/s    in 0s

2022-11-15 19:27:45 (75.2 MB/s) - written to stdout [2765/2765]

Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
mkdir: can't create directory '/opt/bin': No such file or directory
mkdir: can't create directory '/opt/etc': No such file or directory
mkdir: can't create directory '/opt/lib/': No such file or directory
mkdir: can't create directory '/opt/tmp': No such file or directory
mkdir: can't create directory '/opt/var/': Read-only file system
Info: Opkg package manager deployment...
/opt/bin/opkg: No such file or directory
chmod: /opt/bin/opkg: No such file or directory
/opt/etc/opkg.conf: No such file or directory
Info: Basic packages installation...
sh: /opt/bin/opkg: not found
sh: /opt/bin/opkg: not found
chmod: /opt/tmp: No such file or directory
ln: /opt/etc/passwd: No such file or directory
ln: /opt/etc/group: No such file or directory
ln: /opt/etc/shadow: No such file or directory
ln: /opt/etc/gshadow: No such file or directory
ln: /opt/etc/localtime: No such file or directory
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to $PATH variable
Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
ozone@RT-AX58U-8530:/tmp/home/root# opkg
-sh: opkg: not found
ozone@RT-AX58U-8530:/tmp/home/root#

zyxmon, 2022-11-16:
Your /opt folder is not writable!

closed 2022-11-16 by zyxmon

zyxmon, 2022-11-16:
My AX58U runs Entware for 4 years. This guide was used to install -https://github.com/Entware/Entware/wiki/Install-on-Asus-stock-firmware
My current firmware is 3.0.0.4.382_52504-g583f257

kingozone, 2022-11-17:
I have followed the guide step by step and still cannot get the /opt folder to be writable, so I'm not exactly sure what is going wrong here.