Entware/Entware #869: i suggest "unar"

wood9205, 2022-11-25:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.
unzip & 7zip messup with archive have chinese filename. 
i use unzip and 7zip in armv7sf-k3.2
unar deal with this perfectly.
URL: https://unrar.org/

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2022-11-25:
We've got one from https://www.rarlab.com/
Did you tried it?

wood9205, 2022-12-03:
sorry ,the URL: https://unrar.org/  are wrong
it should be URL: https://theunarchiver.com/command-line

The-BB, 2025-06-11:
This package needs a toolchain with Objective-C compiler support.

closed 2025-06-11 by The-BB