Entware/Entware #882: ipset v7.16: Cannot open session to kernel.

olsh, 2022-12-29:
**Describe the bug**
When I can the following command
```
ipset create mylist hash:ip
```

I get the error mentioned in the title
```
ipset v7.16: Cannot open session to kernel.
```

**Environment (please complete the following information):**
 - Entware feed: mipselsf-k3.4
 - Device: Zyxel Keenetic Extra
 - Device firmware: v2.08(AANS.4)C2

I've also installed all the additional modules for the router
[image]

ryzhovau, 2022-12-29:
Looks like kernel is too old,
Just checked thatr one https://keenetic-gi.ga/2022/12/17/adguardhome-and-selective-routing.html on mipsel and aarch64 with no issues.
```
~ # opkg install ipset
...
~ # ipset -V
ipset v7.16, protocol version: 7
Warning: Kernel support protocol versions 6-6 while userspace supports protocol versions 6-7
~ # ipset create myipset hash:ip
~ # uname -a
Linux chewbacca 4.9-ndm-5 #0 SMP Thu Dec 22 16:05:14 2022 mips GNU/Linux
```

closed 2022-12-29 by ryzhovau

olsh, 2022-12-29:
@ryzhovau thanks for the quick answer.
Are there any workarounds for this? Probably there is a way to install an older version of ipset?

ryzhovau, 2022-12-29:
You may find any version of ipset ever been compiled for mipsel here http://bin.entware.net/mipselsf-k3.4/archive/

olsh, 2022-12-29:
Welp, it looks like I have a super old router :)
The oldest version shows the same error:
```
ipset create myipset hash:ip
ipset v6.34: Cannot open session to kernel.
```

Thank you anyway! 👍🏻