Entware/Entware #883: man-db (2.11.2-1): man binary segfaults (x86-64, QNAP)

ktetzlaff, 2022-12-29:
**Describe the bug**
[man-db 2.11.2-1]: A simple `man man` just crashes with a segfault. 

The `man` binary runs fine with `--version` or `--help` options. But as soon as it is trying to locate or display a man page, it segfaults.

Note: Before running `man` as non-root user, there is also an issue with the permissions of `/opt/etc/man_db.conf` which need to be changed from 0600 to e.g. 0644.

When I tried to debug the crash by rebuilding the package locally, the problem disappeared (i.e. the man binary installed from a locally built `man-db_2.11.1-1_x64-3.2.ipk` runs without a problem).

What I noticed: There is a size difference between the `man` binaries:
- man installed from http://bin.entware.net/x64-k3.2/man-db_2.11.1-1_x64-3.2.ipk 108960 bytes
- man installed from locally built man-db_2.11.1-1_x64-3.2.ipk: 117472 bytes

**Environment (please complete the following information):**
 - Entware feed: x64-k3.2
 - Device: QNAP TS-870 Pro
 - Firmware version: QTS 4.3.6 (20221124)

zyxmon, 2023-01-02:
Confirmed

ryzhovau, 2023-06-02:
Done! The newer version will be deployed soon.
https://github.com/Entware/rtndev/commit/b16ffdb452f6280e9673b193bb94b3182cb41072

closed 2023-06-02 by ryzhovau