Entware/Entware #884: yt-dlp not working

bloksma, 2023-01-02:
yt-dlp_2022.9.1-1 not weorking. 

Traceback (most recent call last):
  File "/opt/bin/yt-dlp", line 33, in <module>
    sys.exit(load_entry_point('yt-dlp==2022.9.1', 'console_scripts', 'yt-dlp')())
  File "/opt/bin/yt-dlp", line 25, in importlib_load_entry_point
    return next(matches).load()

Environment:
 - Entware feed: armv7sf-k2.6
 - Device: Zyxel NAS326

zyxmon, 2023-01-02:
Some python packages are missing in dependencies. Wi will fix it later. Just run to fix mannually
```
opkg install python3-logging
opkg install python3-uuid
```

closed 2023-01-02 by zyxmon

bloksma, 2023-01-02:
Package python3-logging (3.10.7-1) installed in root is up to date.
Package python3-uuid (3.10.7-1) installed in root is up to date.

I've tries to use --autoremove and reinstall python3 and yt-dlp, still not working.

zyxmon, 2023-01-02:
It works on my hardware. Please show the full error log.

```
~ # uname -a
Linux ws880 2.6.36.4brcmarm #1 SMP PREEMPT Wed Jun 17 15:39:25 MSK 2015 armv7l GNU/Linux
~ # yt-dlp

Usage: yt-dlp [OPTIONS] URL [URL...]

yt-dlp: error: You must provide at least one URL.
Type yt-dlp --help to see a list of all options.
```

bloksma, 2023-01-02:
~ # uname -a
Linux NAS326 3.10.39 #1 Wed May 11 14:14:53 CST 2022 armv7l GNU/Linux
~ # yt-dlp
Traceback (most recent call last):
  File "/opt/bin/yt-dlp", line 33, in <module>
    sys.exit(load_entry_point('yt-dlp==2022.9.1', 'console_scripts', 'yt-dlp')())
  File "/opt/bin/yt-dlp", line 25, in importlib_load_entry_point
    return next(matches).load()
StopIteration
~ #

zyxmon, 2023-01-02:
You can remove Entware package and try to install yt-dlp manually with pip. Read here
https://github.com/yt-dlp/yt-dlp/wiki/Installation#with-pip

zyxmon, 2023-01-02:
and here
https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules