Entware/Entware #887: Busybox from entware breaks opkg downloads

pelepelin, 2023-01-17:
**Describe the bug**
Busybox from entware breaks opkg downloads. Looks like busybox-wget does not support https.

**Environment (please complete the following information):**
 - Entware feed: ??? aarch64-k3.10
 - Device ASUS RT-AX86U
 - Firmware version ASUSWRT-Merlin RT-AX86U 388.1_0

As shown in the log below, sed (or any other package) is not installable if busybox is installed.

```
# opkg install sed
Installing sed (4.9-1) to root...
Downloading https://bin.entware.net/aarch64-k3.10/sed_4.9-1_aarch64-3.10.ipk
Configuring sed.
# opkg remove sed
Removing package sed from root...
# opkg install busybox
Installing busybox (1.35.0-3) to root...
Downloading https://bin.entware.net/aarch64-k3.10/busybox_1.35.0-3_aarch64-3.10.ipk
Configuring busybox.

[ snip ]

# opkg install sed
Installing sed (4.9-1) to root...
Downloading https://bin.entware.net/aarch64-k3.10/sed_4.9-1_aarch64-3.10.ipk
wget: not an http or ftp url: https://bin.entware.net/aarch64-k3.10/sed_4.9-1_aarch64-3.10.ipk
Collected errors:
 * opkg_download: Failed to download https://bin.entware.net/aarch64-k3.10/sed_4.9-1_aarch64-3.10.ipk, wget returned 1.
 * opkg_install_pkg: Failed to download sed. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package sed.
# opkg remove --force-removal-of-essential-packages busybox
Removing essential package busybox under your coercion.
        If your system breaks, you get to keep both pieces
Removing package busybox from root...
# opkg install sed
Installing sed (4.9-1) to root...
Downloading https://bin.entware.net/aarch64-k3.10/sed_4.9-1_aarch64-3.10.ipk
Configuring sed.
#
```

zyxmon, 2023-01-18:
`Downloading https://bin.entware.net/aarch64-k3.10/sed_4.9-1_aarch64-3.10.ipk``

`opkg.conf` file does not have https links. You've modified (or some installation script not from Entware) it and wget applet from busybox cannot download https. temporary switch to http, run `opkg install wget-ssl`, switch back to https.
You can read more discussion on using https in Entware here in Issues and other places.
PS you can also just delete wget applet (Entware busybox) and not install wget from Entware.

pelepelin, 2023-01-19:
Interesting. I definitely didn't touch `opkg.conf` and didn't run any scripts from Merlin's `amtm` except of entware installation.
Of course, first line of `opkg.conf` is
```
src/gz entware https://bin.entware.net/aarch64-k3.10
```
Might it be an inconsistency of amtm entware script with "no https" policy?

pelepelin, 2023-01-19:
There's something in here

https://github.com/decoderman/amtm/blob/e03fe5317a03d47fa0fe92aecb97dad7a10611d9/amtm_modules/entware.mod#L16

and here

https://github.com/decoderman/amtm/blob/e03fe5317a03d47fa0fe92aecb97dad7a10611d9/amtm_modules/entware_setup.mod#L51

@decoderman FYI

zyxmon, 2023-01-19:
Entware installation script must be a part of Merlin's firmware. Everything will work correctly until you replace wget with a version that does not support https. Just remove /opt/bin/wget symlink - this will fix https problem.
Original Entware scripts (and щзлпюсщта) are here - https://bin.entware.net/aarch64-k3.10/installer/

closed 2023-01-19 by zyxmon

zyxmon, 2023-01-19:
This is not a great idea to use wget-ssl package. In case openssl is updated (it is updated before wget-ssl) wget-ssl stops to work and cannot update itself.
One more option

- install Entware using amtm and change https to http in opkg.conf )))

pelepelin, 2023-01-19:
> In case openssl is updated (it is updated before wget-ssl) wget-ssl stops to work and cannot update itself.

Then it should be noted on https://github.com/Entware/Entware/wiki/Using-HTTPS-with-opkg