Entware/Entware #888: gcc

flowercow, 2023-01-18:


zyxmon, 2023-01-18:
Try to run gcc_env.sh script before compiling.
Please specify your Entware feed (architecture) when posting issues.

flowercow, 2023-01-18:
i run gcc_env.sh or source it before run gcc, no help, i got the same error.
Architecture is armv7sf-k2.6.

zyxmon, 2023-01-18:
Probably your firmware is not compatible. Cannot confirm. The output is form tomato arm router:

```
~ # opkg install gcc
Installing gcc (8.4.0-2) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/gcc_8.4.0-2_armv7-2.6.ipk
Installing binutils (2.38-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/binutils_2.38-1_armv7-2.6.ipk
Installing libzstd (1.5.2-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libzstd_1.5.2-1_armv7-2.6.ipk
Configuring binutils.
Configuring libzstd.
Configuring gcc.
There are no *-dev packages in Entware(with few exceptions)!
Please install headers as described in the wiki:
https://github.com/Entware/Entware/wiki
root@ws880:~$ cat hello.c
#include <stdio.h>

int main()
{
    printf("%s", "Hello\n");
    return 0;
}
root@ws880:~$ gcc hello.c -o hello
root@ws880:~$ ./hello
Hello
```

closed 2023-01-18 by zyxmon

flowercow, 2023-01-21:
after install gcc with opkg, I can compile the hello.c file and output "Hello", but after several hours, gcc output ld error.
i have a model with asuswrt 3.0.0.4.386.9

flowercow, 2023-01-21:
after reinstall gcc, i run it in the terminal, i got these message:
/opt/bin/ld: /tmp/mnt/star/entware/bin/../lib/gcc/arm-openwrt-linux-gnueabi/8.4.0/crt1.o: in function `_start':
(.text+0x34): undefined reference to `main'
collect2: error: ld returned 1 exit status

but i can compile c files as discribed above.