Entware/Entware #889: Neovim

pedropombeiro, 2023-01-19:
**Describe the solution you'd like**
https://neovim.io/ is an improved version of Vim, with more responsive UI, a faster scripting language (Luascript), and built-in Language Server Protocol support. Currently it is possible to install neovim with asdf, but only up to version 0.7. This is a problem, because a few of the most high-value plugins require Neovim 0.8.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

pedropombeiro, 2023-06-02:
@ryzhovau how will I be able to test it? Is it just a matter of waiting a while and running `opkg update && opkg install neovim`?

ryzhovau, 2023-06-02:
The second. It will be available soon as a ordinary part of repo.

pedropombeiro, 2023-06-03:
Thanks. I just installed, and it seems to be working fine! Only a minor warning at the end of the installation:

```
 opkg install neovim
Installing neovim (0.9.0-1) to root...
Downloading http://bin.entware.net/x64-k3.2/neovim_0.9.0-1_x64-3.2.ipk
Installing luajit (2.1.0-beta3-7) to root...
Downloading http://bin.entware.net/x64-k3.2/luajit_2.1.0-beta3-7_x64-3.2.ipk
Installing luv (1.43.0-0-1) to root...
Downloading http://bin.entware.net/x64-k3.2/luv_1.43.0-0-1_x64-3.2.ipk
Installing msgpack-c (6.0.0-1) to root...
Downloading http://bin.entware.net/x64-k3.2/msgpack-c_6.0.0-1_x64-3.2.ipk
Configuring msgpack-c.
Configuring luajit.
Configuring luv.
Configuring neovim.
Generating tags...
Error detected while processing command line:
E154: Duplicate tag "cscope" in file /opt/share/nvim/runtime/doc/vim_diff.txt
```