Entware/Entware #893: [Package Request] mosh

u2324, 2023-01-28:
Mosh enables SSH sessions to function while roaming between networks.  Instructions are given on building opkg files, but it's not in the repo.

https://github.com/mobile-shell/mosh/wiki/Build-Instructions#openwrt

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. x86_64] and get a feedback about it.

token0, 2023-02-08:
How did you search?

```

mosh-client - 1.4.0-1 - Mosh is a remote terminal application that allows roaming, supports intermittent connectivity, and provides intelligent local echo and line editing of user keystrokes. This is a mosh client.

mosh-full - 1.4.0-1 - Mosh is a remote terminal application that allows roaming, supports intermittent connectivity, and provides intelligent local echo and line editing of user keystrokes. This is a full package with perl wrapper script

mosh-server - 1.4.0-1 - Mosh is a remote terminal application that allows roaming, supports intermittent connectivity, and provides intelligent local echo and line editing of user keystrokes. This is a mosh server.

```

u2324, 2023-02-08:
I searched using the `opkg find` command with the string `"mosh"` and the regular expression `".*mosh.*"`.

There are 2687 packages in the repos, so I figured I had that set up okay.

If I do `opkg list | grep mosh`, then the results are shown properly, as you've given them.

Thanks for your help!

```
sh-4.4# opkg -h
[...]
	find <regexp>		List packages whose name or description matches <regexp>
[...]
sh-4.4# opkg find "mosh"
sh-4.4# opkg find ".*mosh.*"
sh-4.4# opkg list | wc -l
2687
sh-4.4# opkg list | grep mosh
mosh-client - 1.4.0-1 - Mosh is a remote terminal application that allows roaming, supports intermittent connectivity, and provides intelligent local echo and line editing of user keystrokes. This is a mosh client.
mosh-full - 1.4.0-1 - Mosh is a remote terminal application that allows roaming, supports intermittent connectivity, and provides intelligent local echo and line editing of user keystrokes. This is a full package with perl wrapper script
mosh-server - 1.4.0-1 - Mosh is a remote terminal application that allows roaming, supports intermittent connectivity, and provides intelligent local echo and line editing of user keystrokes. This is a mosh server.

```

closed 2023-05-23 by ryzhovau