Entware/Entware #900: Pyload 0.4.20

geronimoronin, 2023-02-08:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

please add pyload, since last update is no longer included.
https://github.com/Entware/Entware-ng/wiki/Using-Pyload
https://github.com/pyload/pyload/tree/stable

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),
- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

snipah19, 2023-05-04:
+1 i cannot find pyload on entware list on my armv5 TS-219p following the steps on [Using-Pyload](https://github.com/Entware/Entware-ng/wiki/Using-Pyload)

[~] # opkg install pyload duktape python-crypto python-imglib tesseract                                              
Unknown package 'pyload'.
Package duktape (2.7.0-1) installed in root is up to date.
Unknown package 'python-crypto'.
Unknown package 'python-imglib'.
Package tesseract (4.1.1-4) installed in root is up to date.
Collected errors:
 * opkg_install_cmd: Cannot install package pyload.
 * opkg_install_cmd: Cannot install package python-crypto.
 * opkg_install_cmd: Cannot install package python-imglib.

ryzhovau, 2023-05-23:
Entware-ng is EoL.

closed 2023-05-23 by ryzhovau