Entware/Entware #901: qbittorrent CPU load issues by web UI

YoshiKill, 2023-02-09:
Greating, I update qbittorrent_4.3.9-1_mipsel-3.4.ipk package that works fine, to latest version qbittorrent_4.4.5-1_mipsel-3.4.ipk, and with this version i have an issues. 

In this qbittorrent_4.4.5-1_mipsel-3.4 version I find that qbittorrent web UI process load CPU 50-60% in background, even you don't use web UI or torrent don't load anything. 
Htop to show problem 
[image]

Entware: mipselsf-k3.4
Device: Lenovo newifi d2
Firmware: Padavan 3.4.3.9L-102_06dd348

zyxmon, 2023-02-09:
Cannot confirm it. The screenshot is from similar mipsel router.
[image]

zyxmon, 2023-02-09:
Удалось повторить, после захода в настройки.
Это баг. Поищите на сайте qtorrent. Скорее всего в новой версии вылечили. Но собрать ее не смогли пока.

The-BB, 2023-02-09:
http://bin.entware.net/mipselsf-k3.4/test/libtorrent-rasterbar_2.0.8-1_mipsel-3.4.ipk
http://bin.entware.net/mipselsf-k3.4/test/qbittorrent_4.5.0-1_mipsel-3.4.ipk

YoshiKill, 2023-02-09:
> http://bin.entware.net/mipselsf-k3.4/test/libtorrent-rasterbar_2.0.8-1_mipsel-3.4.ipk
> http://bin.entware.net/mipselsf-k3.4/test/qbittorrent_4.5.0-1_mipsel-3.4.ipk

Благодарю, как раз по вашей инструкции с форума Кеенетик уже собирался учиться собирать. К сожаления в новой версии 4.5.0-1 проблема с загрузкой процессора в холостую осталась. Теперь разброс от нескольких процентов до 100% с абсолютным рандомом этой самой нагрузки. Возможно виной всему мой старый конфиг и потому что я обновлялся, попробую сделать чистую установку, ради эксперемента. 

> Это баг. Поищите на сайте qtorrent. Скорее всего в новой версии вылечили.

увы проблема не ушла и решения или даже вопросов в ветке с багами об тоже не нашёл. Стоит туда об этом писать?

zyxmon, 2023-02-09:
Я сделал чистую установку - стало не сильно лучше. Похоже qbittorrent не для mips* железок.

YoshiKill, 2023-02-09:
> Похоже qbittorrent не для mips* железок.

но старая версия 4.3.9-1 работала без нареканий и ничего не грузила в фоне, видимо намудрили с оптимизацией, оно и не мудрено, новый клиент даже на пк у меня стал дико тормозным со своих обновленим, при то что старая версия летала. Ну и дизайн тоже стал спорный. А есть какие то альтернативы в хороших клиентах для роутера? Трансмишн работает тоже так себе, часто не грузит ничего, не видит сиды и тупит. 

Снёс кстати папку с настройкми _/opt//etc/qBittorrent_  версия 4.5 перестала грузить процессор и как либо отсвечивать в активных процессах, правда клиент абсолютно голый. Попробую, что-то скчать и проиндексировать старые торренты, чтоб посмотреть как оно будет в работе себя вести.

zyxmon, 2023-02-09:
deluge использует ту жк либу (движок), что и qbittorrent.

YoshiKill, 2023-02-11:
> deluge использует ту жк либу (движок), что и qbittorrent.

к сожалению почему то он у меня не запустился 

Отписываюсь о своих эксперементах с последней qbt 4.5 версией, после чистой установки с удаленим всех конфигов как и писал, клиент перестал постоянно грузить процессор на 50-60%. 
Пришлось заново дабавлять все торенты в ручную, многие к сожалению почему то не хотели индексироваться и даже при наличии файлов. Удалось востановить только часть торрентов к сожалению.  Всего 30 штук. Все скачанные без ошибок и тд. После этого клиент начал загружать процессор в фоне с краткосрочными прыжками 2-4-7% иногда редко подскакивает до 20%. В целом можно сказать что ситуация нормализовалась. И баг или не оптимизация нового клиента с перегрузкой процессора происходит либо обновлении со старой версии где происходит либо конфликт либо не совместимость со старыми конфигами и раздачами. Либо же чем больше активных торентов тем сильнее новый клиент загружает процессор. Напомню в версии 4.3.9. таких проблем не было, клиент в фоне вообще не потреблял ресурсы, даже в пару процентов, пока не начиналась какая то скачка.

ryzhovau, 2023-05-23:
Ну и отлично!

closed 2023-05-23 by ryzhovau