Entware/Entware #904: [Package Request] HSTR bash command history utility

mrjnet, 2023-02-20:
HSTR (HiSToRy -- https://github.com/dvorka/hstr) is used to view, navigate, search (even with regular expressions) and set favourite commands in the bash command history.  I have found this invaluable when developing scripts and general system maintenance & troubleshooting.  On my routers, using the std. bash history search is limited and more cumbersome than this utility.

- [X] No similar package already added and/or this one got some new killer feature(s)

- [X] Package is in active development (at least, not abandoned for 5 years or more),
       >>> Latest release is 11Dec2022
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.
       >>> I'm able to test on aarch64-k3.10

ryzhovau, 2023-06-02:
Done. Binaries will be deployed soon.

closed 2023-06-02 by ryzhovau

mrjnet, 2023-06-04:
Thx very much.  I've test it and it works great.