Entware/Entware #912: Qbittorrent - nox (entware) issue with cyrillic symbols

YoshiKill, 2023-03-09:
Grating. I have problem with downloading torrent file that has Cyrillic symbols in name, when i start download it became en error file. 
I use qbitt on my router from [[](url)](https://bin.entware.net/mipselsf-k3.4/) package
entware - mipselsf-k3.4
Newifi D2 Padavan [3.4.3.9L-102_c02fd39]
qbittorrent - 4.5.0-1 
libtorrent-rasterbar - 2.0.8-1
qt5-core - 5.15.7-1

i found this topic with same problem https://github.com/qbittorrent/qBittorrent/issues/14837

author say that problem is with 
https://github.com/qbittorrent/qBittorrent/issues/14837#issuecomment-826309571
```
I found the problem.
Setting LC_* envs to something like *.UTF-8 solves the problem described above with libtorrent v2.0.3. I think it's right solution.
libtorrent v1.2.13 works well with LC_* envs set to POSIX.
```
any one know how fix this issues with this entware version? 

error screen
[image]

logs 

```
(W) 2023-03-09T16:29:06 - Предупреждение об ошибке файла. Торрент: «Odium / Gorky-17) [L] [RUS] (1999) [1С]». Файл: «/media/AiDisk_b2/transmission/qbtorrent/Game/ÐÐ¾ÑÑÐºÐ¸Ð¹-18 - ÐÑÐ¶ÑÐºÐ°Ñ ÑÐ°Ð±Ð¾ÑÐ°/Gorkiy18.1999.PC.ISO». Причина: «Горький-18 - Мужская работа file_open (/media/AiDisk_b2/transmission/qbtorrent/Game/Горький-18 - Мужская работа/Gorkiy18.1999.PC.ISO) error: No such file or directory»
(W) 2023-03-09T16:29:32 - Предупреждение об ошибке файла. Торрент: «Gorky-17». Файл: «/media/AiDisk_b2/transmission/qbtorrent/Game/ÐÐ¾ÑÑÐºÐ¸Ð¹-18 - ÐÑÐ¶ÑÐºÐ°Ñ ÑÐ°Ð±Ð¾ÑÐ°/Gorkiy18.1999.PC.ISO». Причина: «Горький-18 - Мужская работа file_open (/media/AiDisk_b2/transmission/qbtorrent/Game/Горький-18 - Мужская работа/Gorkiy18.1999.PC.ISO) error: No such file or directory»
(W) 2023-03-09T16:30:06 - Предупреждение об ошибке файла. Торрент: «Gorky-17». Файл: «/media/AiDisk_b2/transmission/qbtorrent/Game/ÐÐ¾ÑÑÐºÐ¸Ð¹-18 - ÐÑÐ¶ÑÐºÐ°Ñ ÑÐ°Ð±Ð¾ÑÐ°/Gorkiy18.1999.PC.ISO». Причина: «Горький-18 - Мужская работа file_open (/media/AiDisk_b2/transmission/qbtorrent/Game/Горький-18 - Мужская работа/Gorkiy18.1999.PC.ISO) error: No such file or directory»
(N) 2023-03-09T16:30:30 - Торрент удалён вместе с его содержимым. Торрент: «Gorky-17»
(N) 2023-03-09T16:30:48 - Добавлен новый торрент. Торрент: «Горький-18 - Мужская работа»
(W) 2023-03-09T16:30:53 - Предупреждение об ошибке файла. Торрент: «Горький-18 - Мужская работа». Файл: «/media/AiDisk_b2/transmission/qbtorrent/Game/ÐÐ¾ÑÑÐºÐ¸Ð¹-18 - ÐÑÐ¶ÑÐºÐ°Ñ ÑÐ°Ð±Ð¾ÑÐ°/Gorkiy18.1999.PC.ISO». Причина: «Горький-18 - Мужская работа file_open (/media/AiDisk_b2/transmission/qbtorrent/Game/Горький-18 - Мужская работа/Gorkiy18.1999.PC.ISO) error: No such file or directory»

```

zyxmon, 2023-03-09:
This is similar to https://github.com/Entware/Entware/issues/830

zyxmon, 2023-03-09:
The current version in repo is different
```
~ # opkg list qbit*
qbittorrent - 4.4.5-1 - qBittorrent is a bittorrent client programmed in C++ / Qt that uses
 libtorrent (sometimes called libtorrent-rasterbar) by Arvid Norberg.
 It aims to be a good alternative to all other bittorrent clients out
 there. qBittorrent is fast, stable and provides unicode support as
 well as many features.
```

closed 2023-03-09 by zyxmon

YoshiKill, 2023-03-09:
> The current version in repo is different

знаю, это пакет из тестовой версии, тот что 4.4.5-1 я уже сюда отписал с проблемой, что он сильно нагружает роутер. 4.5 версия не так жрёт ресурсы, но всплыла вот эта проблема с руганью на локаль.

zyxmon, 2023-03-09:
https://github.com/Entware/rtndev/blob/master/qbittorrent/files/S89qbittorrent#L9
эта строка есть?

YoshiKill, 2023-03-09:
> https://github.com/Entware/rtndev/blob/master/qbittorrent/files/S89qbittorrent#L9
> эта строка есть?

Спасибо, добавил строчку и заработало без ошибок. Хорошо бы эту строчку по умолчанию добавить в пакет. Или она там есть, это в тестовом без неё?

zyxmon, 2023-03-09:
В 4.5.2 будет.

zyxmon, 2023-07-11:
Да - по ошибке выкинули нужную строку - верните руками. А еще лучше не использовать неправильные torrent файлы с кривой кодировкой.

JugglerLKR, 2023-07-11:
> верните руками.

export LC_ALL=en_US.UTF-8
это?
или же ru_RU.UTF-8?

zyxmon, 2023-07-11:
https://github.com/Entware/rtndev/blob/master/qbittorrent/files/S89qbittorrent