Entware/Entware #913: OpenSSL 3.0 breaks packages

multiduplikator, 2023-03-15:
**Describe the bug**
After latest update  in x64-k3.2, openssl got upgraded to 3.0 branch. At least weechat, rtorrent and curl are throwing error messages:

"error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory"

Seems like they did not get built against the new openssl.

**Environment (please complete the following information):**
 - Entware feed: x64-k3.2
 - Device: Qnap TS-470pro
 - Firmware version 4.3.6.2232

zyxmon, 2023-03-16:
Opkg is too straightforward and simple. Try this - https://github.com/Entware/Entware/wiki#%EF%B8%8F-package-upgrade-notes
If it does not help you can try something like 
`opkg list-installed|grep -v "libc "|sed 's/ - .*$//'|grep lib|grep -v libpthread|grep -v libgcc|xargs -n 5 opkg --force-reinstall install`

multiduplikator, 2023-03-16:
Added a ^ before the grep for lib. That worked. Now having issues with some upgrade versions mismatch...Will deal with that. But the issue is resolved

opkg list-installed|grep -v "libc "|sed 's/ - .*$//'|grep ^lib|grep -v libpthread|grep -v libgcc|xargs -n 5 opkg --force-reinstall install

multiduplikator, 2023-03-16:
Had to edit the install status file manually and remove libattr and libx264 (in my case). After that, all is back to normal.

OneCDOnly, 2023-03-16:
I think I'm getting the same thing with the Deluge packages (`deluge` & `deluge-ui-web`). 

Here's the Deluge daemon startup:
```
'/opt/bin/deluged -L info --logfile /share/CACHEDEV2_DATA/.qpkg/Deluge-server/config/Deluge-server.log --config /share/CACHEDEV2_DATA/.qpkg/Deluge-server/config/ --pidfile /var/run/Deluge-server.pid'
OK
file /var/run/Deluge-server.pid: exists
Traceback (most recent call last):
  File "/opt/bin/deluged", line 33, in <module>
    sys.exit(load_entry_point('deluge==2.1.1', 'gui_scripts', 'deluged')())
  File "/opt/lib/python3.10/site-packages/deluge/core/daemon_entry.py", line 87, in start_daemon
    from deluge.core.daemon import is_daemon_running
  File "/opt/lib/python3.10/site-packages/deluge/core/daemon.py", line 19, in <module>
    from deluge.core.core import Core
  File "/opt/lib/python3.10/site-packages/deluge/core/core.py", line 40, in <module>
    from deluge.core.rpcserver import export
  File "/opt/lib/python3.10/site-packages/deluge/core/rpcserver.py", line 28, in <module>
    from deluge.crypto_utils import check_ssl_keys, get_context_factory
  File "/opt/lib/python3.10/site-packages/deluge/crypto_utils.py", line 12, in <module>
    from OpenSSL import crypto
  File "/opt/lib/python3.10/site-packages/OpenSSL/__init__.py", line 8, in <module>
  File "/opt/lib/python3.10/site-packages/OpenSSL/crypto.py", line 17, in <module>
  File "/opt/lib/python3.10/site-packages/OpenSSL/_util.py", line 6, in <module>
  File "/opt/lib/python3.10/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 14, in <module>
ImportError: /opt/lib/python3.10/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so: undefined symbol: FIPS_mode
```
I've reinstalled all libs as shown earlier in this issue, I've reinstall `python*`, I even resorted to reinstalling Entware entirely, but I still get this error with OpenSSL. Is Deluge now broken?

zyxmon, 2023-03-17:
It looks like deluge is broken.

zyxmon, 2023-03-17:
@OneCDOnly 
Please check if the deluge is working. It is better to make full Entware reinstall
Fixed via
https://github.com/Entware/entware-packages/commit/31802a768b377cdc7263553051163ac33163af58
and
https://github.com/Entware/Entware/commit/8da0e370648fa0bce98e7887dd6951b626d371d4

OneCDOnly, 2023-03-17:
@zyxmon yes, that's fixed it. Cheers mate! 👍🏽

victor-rds, 2023-03-20:
In my case also broke "wget", which in turn broke opkg:
```
$ opkg update
Downloading http://bin.entware.net/x64-k3.2/Packages.gz
wget: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
*** Failed to download the package list from http://bin.entware.net/x64-k3.2/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/x64-k3.2/Packages.gz, wget returned 127.
```
Removing the wget package fixes opkg, but broke my scripts that use the full wget.

ryzhovau, 2023-03-20:
I'll keep this one open for few days, while this topic stays hot.

YujiaCheng1996, 2023-03-25:
nginx-extras breaked! My website is down now...

crkinard, 2023-03-28:
git is suffering the same issues as well.

zyxmon, 2023-04-04:
> copy the two files llibcrypto.so.1.1 and libssl.so.1.1 to /opt/lib can fix this issue temporarily.

There are correct fixes in this discussion. Do not use incorrect ones.

zyxmon, 2023-04-04:
@gilgrissom - what workarounds you have tried? What were your results?

Weegley, 2023-04-09:
curl - 7.86.0-1 also broken

-bash-5.2# curl
curl: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory

original-birdman, 2023-04-19:
> In my case also broke "wget", which in turn broke opkg:

If you run the upgrade as:
```
PATH=/usr/bin:$PATH opkg upgrade
```
it will use the system `wget`, which let's it download the rest of the things.

Shorrer, 2023-04-23:
Reinstalling everything seemed to help the issue. At least curl works now, so it is something that is worth a try. As said above -- you're probably better off reinstalling entire Entware, but this is not a solution that's going to work with everyone.

So, maybe someone will find information there useful.
**!!! IT IS NOT CHECKED THOROUGHLY !!!** This is more of a list of advices and things to know than a proper guide. Only do this if you're sure what you're doing and how it affects your system. Do a backup that you will be able to restore quickly if you need to.

0. Do a backup of your configuration and stuff (and better yet -- dump the whole filesystem somewhere)
1. Generate a package list
It is better to have this list at hand somewhere in file because if things didn't go the way they should you'd at least have list of installed packages. What seems to be an OK command for me is 
`opkg list-installed | awk '!/^(busybox|opkg|lib(c|pthread|gcc)) / { print $1 }'`
2. Create package cache
It is very easy to lose ability to download packages when doing the reinstall. If you have a package cache then all you have to have is `busybox` (without it opkg will not be able to install anything), and if you didn't remove `libc` and `libgcc` it is possible to just copy compatible `busybox` binary to `/opt/bin` and create symlinks.
You can do this by 
`cd /path/to/package/cache && opkg list-installed | awk '{ print $1 }' | xargs opkg download && cd -`
3. Reinstall the world
`opkg list-installed | awk '!/^(busybox|opkg|lib(c|pthread|gcc)) / { print $1 }' | xargs opkg --cache=/path/to/package/cache install`

If things go wrong:
* `libgcc` and `libc` are essential for sure. `libpthread` is too, but I'm not sure. If you don't have them you're on your own -- full reinstall might actually be the best solution at this point if you're in a hurry.
* If you have the above packages then it's recoverable. You basically want to restore `opkg`, `wget` and `busybox` if you've lost them. Hopefully you have made the package cache -- for some reason I can't install `libopenssl` without it as it always wants to download even when I give OPKG the filename. If not, then you can create an empty directory that you will use as package cache and place `*.ipk` files in there.
* You might've lost the `ca-bundle`. This is not a huge problem, but if you have to download anything `wget` will fail and `opkg` will report that it failed. `--no-check-certificate` is a solution there.
* If you've lost the Busybox then OPKG doesn't want to install anything (and it's a pain to lookup directory contents through shell completion). As long as you have essential libraries mentioned above you can grab a compatible `busybox` binary (e.g. from Entware install archive) and drop it into `/opt/bin/` (or wherever you have your Entware installed). Depending on the way you do it you might end up with executable without executable permissions, and if you've lost the Busybox you've probably lost the `chmod` as well. It is not a problem: try `/opt/lib/ld-2.27.so /opt/bin/busybox chmod +x /opt/bin/busybox` (your `ld` might have some other version). It, apparently, ignores the fact that there are no executable permissions on the file. Then it's the case of creating the symlinks:
`cd /opt/bin && /opt/bin/busybox --list | /opt/bin/busybox xargs -n1 /opt/bin/busybox ln -s /opt/bin/busybox && cd -`
After all this you should have a bunch of symlinks to Busybox in `/opt/bin/` and a useable system. Don't forget to install Busybox using OPKG so that you would continue to get updates to it (and to do whatever postinstall script might want to do).
* If you've lost OPKG then you should also be able to just copy it.

Hopefully it'll help someone.

closed 2023-06-02 by ryzhovau